platformio/platformio-vscode-ide #1420: Installation Manager

WesleyJoseSantos, 2019-12-03:
Hello! Pio it was working fine for me, but for some unknown reason the project tasks has broken, givin me the message:  no task to run found. When a tried to reinstall, an exception it ocourred:

  %23 Configuration

  VSCode: 1.40.2
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000005d4 (most recent call first):

    at t.value (c:\Users\BINAR\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\installer\stages\platformio-core.js:302:17)
  ```

Thanks for the help.

ivankravets, 2020-03-01:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/1570#issuecomment-593128668? Does it work for you?

WesleyJoseSantos, 2020-03-03:
Thanks! I was using pio CLI on vscode terminal to work. I will try your solution at end of week.

WesleyJoseSantos, 2020-03-12:
> Could you try this solution [#1570 (comment)](https://github.com/platformio/platformio-vscode-ide/issues/1570#issuecomment-593128668)? Does it work for you?

My problem it solved by following steps: CTRL+Shift+P: Preferences: Open Settings (JSON) Delete "task.autoDetect": "off". Thanks!

closed 2020-03-12 by WesleyJoseSantos