platformio/platformio-vscode-ide #1436: Is this range correct?

pfeerick, 2019-12-11:
https://github.com/platformio/platformio-vscode-ide/blob/f934238fdd8fae4ae1458b5d599f887fd51137e4/package.json#L578

Since PIO Home defaults to port 8008, one would think it would be included in the default 'automatic' search? What determines the range... the built in web browser? Just wondering if it can change, as it seems it [didn't work on the Raspberry Pi 4 for one user](https://community.platformio.org/t/platformio-on-raspbian-rpi4b-stuck-loading/11011/8?u=pfeerick).

ivankravets, 2020-06-03:
No, it's a static port. Not a range. If you keep this field empty, we will search a free port in that range.

pfeerick, 2020-06-04:
Maybe I phrased that poorly. If PIO Home defaults to 8008... why is that not included in the automatic search range(according to that description of the parameter)... since if it searches between 8010..8100 ... 8008 will never be found! 😆 

Since PIO Home usually does actually work, I would suggest that either the default is NOT 8008, or that comment is wrong... at the time I posted this I would have said it was the comment, since the user in question set it to 8008 and was able to access PIO Home, but that is just a guess.

closed 2020-09-18 by stale[bot]