platformio/platformio-vscode-ide #1462: Installation Manager

GuangYiL, 2019-12-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.41.1
  PIO IDE: v1.10.0
  System: Darwin, 19.2.0, x64

  %23 Exception
  ```
  Error: PIP Core: DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/%23python-2-support
WARNING: The repository located at pypi.douban.com is not a trusted or secure host and is being ignored. If this repository is available via HTTPS we recommend you use HTTPS instead, otherwise you may silence this warning and allow it anyway with '--trusted-host pypi.douban.com'.
ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
WARNING: The repository located at pypi.douban.com is not a trusted or secure host and is being ignored. If this repository is available via HTTPS we recommend you use HTTPS instead, otherwise you may silence this warning and allow it anyway with '--trusted-host pypi.douban.com'.

	at /Users/andy/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:35084
	at ChildProcess.d (/Users/andy/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:8215)
	at ChildProcess.emit (events.js:200:13)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-03:
H,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

closed 2020-06-03 by ivankravets

GuangYiL, 2020-06-03:
> H
> 
> 谢谢你的报告！我们从头开始重写了 PlatformIO 核心安装程序。请**下载并安装**[最新的平台IO IDE VSCode 2.0 测试版](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097)，如该页面上**的"试用测试版**"部分所述。
> 
> 请响应此线程，如果它帮助您或您仍然有一个问题。我们将再次重新打开此问题。
> 
> 问候，
> 你的朋友在平台IO

thank you