platformio/platformio-vscode-ide #1469: .vscode json files should use relative and env variables path

stoune, 2020-01-02:
Auto-generated files 
.vscode/c_cpp_properties.json
.vscode/launch.json
contains absolute path, for example:

```
"configurations": [
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            **"executable": "/Users/username/Projects/projname/.pio/build/esp32dev/firmware.elf",
            "toolchainBinDir": "/Users/username/.platformio/packages/toolchain-xtensa32/bin",**
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            },
            "internalConsoleOptions": "openOnSessionStart"
        },

```

The absolute path has two issues:
1. It is not portable across machine
2. It is expose username in the path if I commit them into public git repo

VS Code supports a bunch of [variables](https://code.visualstudio.com/docs/editor/variables-reference)  which should make these path more portable.
${workspaceFolder}  - will be a folder where project resides.
${workspaceFolder}  - the path for the Platformio installation.

stoune, 2020-01-02:
${env.HOME}/.platformio - the path for the Platformio installation.

zfields, 2023-06-07:
3 years!? Come on, guys. This is so easy to fix, and this is such a nuisance.

You must already be using variables to create the file in the first place. Just make them string literals instead of variables and the whole thing works.

**`launch.json`**
```
{
    ...
    "executable": "${workspaceFolder}/.pio/build/<board>/firmware.elf",
    ...
    "toolchainBinDir": "${env:HOME}/.platformio/packages/<toolchain>/bin",
    ...
}

This prevents a team from working on a code base in a single repository. We cannot check in the `.vscode` folder, what an unnecessary annoyance!

R0binBl00d, 2026-10-04:
Still the case on extension 3.3.4 / Core 6.2.0, Windows 11. I work on a
Windows desktop and take a MacBook to the customer, so one repository is
used on both and commits `.vscode/launch.json`, which holds our own launch
entries as well as the three PlatformIO ones, so ignoring the file is not
an option. To make it portable we replaced the generated absolute paths by
hand: `executable` = `${workspaceFolder}/.pio/build/<env>/firmware.elf`
and a home-folder variable for `toolchainBinDir`. Those values work, but
PlatformIO overwrites them with absolute paths on every window reload and
every change of the environment picker, so the file is dirty again and
wrong on the other system.

Could the generated entries use `${workspaceFolder}` for `executable`,
and `${userHome}` (a VS Code variable that exists on every OS;
`${env:HOME}` is not defined on Windows) for `toolchainBinDir` when the
core directory is the default `~/.platformio`? That would make a
committed `launch.json` stable.