platformio/platformio-vscode-ide #1473: Unable to update or install u8g2 library

wisegeeksrules, 2020-01-05:
I'm having  issues trying to update u8g2 library to version 2.27.3. Trying to update gives me a **PIO Core Call Error** and this results in the library no longer appearing under Installed libraries.

The only workaround I have at the moment is to install the previous version of the library which is version 2.27.2

Below is the message I'm getting and the screenshot:

[image]

aderusha, 2020-01-10:
I'm seeing this same behavior here.

wisegeeksrules, 2020-01-12:
> I'm seeing this same behavior here.
> 
> edit: actually that's wrong - I can install no versions of this library that I've tested. Other libraries work as expected. My projects are now failing build as a result.

Looks like the only workaround now is to go to PlatformIO Libraries registry search for U8g2 and install version 2.27.1

aderusha, 2020-01-12:
Thanks @wisegeeksrules!  Can confirm that 2.27.1 does in fact install.

afbayonac, 2020-05-31:
I try install the version 2.27.1 and no works me.

wisegeeksrules, 2020-06-01:
> I try install the version 2.27.1 and no works me.

I'm not sure what causes this, I was able to update to v2.28.1 after setting python 3.x as the default, before I had python 2.7 as the default.
A week later I could not update to the latest version available (v2.28.6) again, (sad). The only work around was to revert to v.2.28.1

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

closed 2020-06-03 by ivankravets

wisegeeksrules, 2020-06-05:
[image]

The issue continues and now I can't install any version of the library, it's not working.

wisegeeksrules, 2020-06-08:
@ivankravets   After giving up, I switched off my machine and 3 days later I tried to install the library starting with u8g2 version 2.28.1 and it worked. then I decided to update to v2.28.6 and it worked again. I'm happy it worked this time.

[image]

[image]

afbayonac, 2020-06-12:
today try again and mysteriously work 😃