platformio/platformio-vscode-ide #1474: command 'platformio-ide.showHome' not found

dcboy, 2020-01-05:
Version: 1.41.1
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T14:57:51.166Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Darwin x64 19.2.0

no have any icon at toolbar and use command+shitf+p open platformio home fail

BCNelson, 2020-02-10:
I'm having this same problem 
Version: 1.42.0
Commit: ae08d5460b5a45169385ff3fd44208f431992451
Date: 2020-02-06T10:51:23.649Z
Electron: 6.1.6
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 5.5.2-arch1-1

bilogic, 2020-04-14:
1.44.0 also...

infinity0n3, 2020-05-18:
Version: 1.45.1
Commit: 5763d909d5f12fe19f215cbfdd29a91c0fa9208a
Date: 2020-05-14T08:27:22.494Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 5.3.0-51-generic

Having the same issue

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

infinity0n3, 2020-06-03:
New Beta didn't help but removing ~/.vscode folder and running vscode again and THEN installing platformIO did resolve the issue.

infinity0n3, 2020-06-03:
Actually, if there is no platformio.ini in the workspace folder the issue appears.
Even an empty platformio.ini file makes it work.

ivankravets, 2020-06-03:
Sorry, you are right, this is a bug.

ivankravets, 2020-06-03:
Could someone provide step by step instruction how to reproduce this issue?

infinity0n3, 2020-06-03:
1. Create a new empty directory
```
>> mkdir test
>> cd test
```
2. open vscode from there 
```
>> code .
```
3. In VSCode go to Command Palette `CTRL+SHIFT+P` and run `PlatformIO: Home`

You get the error
```
Command 'PlatformIO: Home' resulted in an error (command 'platformio-ide.showHome' not found)
```

ivankravets, 2020-06-03:
Do you see HOME icon at the bottom of toolbar? It seems that you disable PlatformIO IDE extension in VSCode settings. If you enable it only for the PlatformIO project, you can't use it when the project is not open.

Did you enable `Activate Only On PlatformIO Project`?

infinity0n3, 2020-06-03:
True. 
I had `Active Only On PlatformIO Project`. 
Now that I turned that setting off I see the HOME icon.

This could have been a the issue from the start.

ivankravets, 2020-06-03:
We can't fix this issue. The problem is with VSCode. The commands are declared in a static file https://github.com/platformio/platformio-vscode-ide/blob/develop/package.json#L92

Even if extension is not active, VSCode shows for user these commands :( As result, you run command but there is no handler for it because extension is not active :(

closed 2020-06-03 by ivankravets

be-engineer, 2020-10-25:
I met same problem,how to solve it?

ivankravets, 2020-10-25:
@be-engineer what is your OS? Do you use the latest VSCode and PlatformIO IDE? Try to remove `%HOME_DIR%/.platformio` folder and restart VSCode.

bilogic, 2020-10-25:
This issue tends to happen if you have not used your VSCode with PlatformIO for a while.
Uninstall and re-install PlatformIO.

ljmerza, 2020-11-06:
reinstalling doesnt work. tried reinstallng vscode, python, platformio. nothing.

ljmerza, 2020-11-06:
unchecking the `Platformio-ide: Activate Only On Platform IOProject`setting and closing out and reopening vscode fixed it for me.

frohro, 2021-04-16:
I had this problem, there was no Platformio Home and I would get the error in VScode even with a platformio project in the editor. 
 and the Platformio-ide: Activate Only On Platform IOProject was not set.  I set it, restarted VScode, and the home appeared after restarting VScode.  I then unset the option, and restarted VScode, and I still have home!

TurnOffNOD, 2022-04-02:
PlatformIO is slow to start, so I need to set "Platformio-ide: Activate Only On Platform IOProject" to true.
Taking this into consideration, how can I bring up PlatformIO home and create a new project?

tionebrr, 2022-11-21:
Any definitive fix for this issue? I'm unable to install platformio locally nor on a ssh remote...

ivankravets, 2022-11-22:
@tionebrr, could you provide any logs?

tionebrr, 2022-11-23:
@ivankravets I had `Activate Only On Platform IOProject` set. I can use Platformio locally now. 
Still not working on a VSCode SSH remote. Will try to share logs later.

tylercamp, 2023-09-15:
Was the field name changed? I don't see anything in VS Code settings containing the text "Activate Only On"

DoubleBoba, 2023-11-18:
@tylercamp 
uncheck Platformio-ide: Auto Open Platform IOIni File