platformio/platformio-vscode-ide #1482: EEPROM.h (sometimes( not found with ezTime lib

vbandke, 2020-01-13:
Probably an error 50 (as in "Error sits 50 cm in front of the screen"), or a bug, or a feature?

Environment:  Win 10 professional, latest build,
Pythion 3.8.1 64 bit
freshly installed PlatformIO, VSCODE IDE

With the Arduino IDE there is no issue.
The situation:  When I compile the sample timezone.cpp of the eztime library, I get a compile error telling me that, while compiling the library, the include file EEPROM.h cannot be found.  

`
Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 2.3.2 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 2.20603.191216 (2.6.3), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Looking for WiFi library in registry
Found: https://platformio.org/lib/show/870/WiFi
LibraryManager: Installing id=870
Using cache: C:\Users\Volker\.platformio\.cache\5a\72673e17beccaf1864b0d3a72b5a185a
WiFi @ 1.2.7 has been successfully installed!
Looking for ezTime library in registry
Found: https://platformio.org/lib/show/5751/ezTime
LibraryManager: Installing id=5751
Using cache: C:\Users\Volker\.platformio\.cache\40\e08f1c651eb6f0293d8be1a4c06a9d40
ezTime @ 0.8.3 has been successfully installed!
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.2.7
|   |-- <SPI> 1.0
|-- <ezTime> 0.8.3
Building in release mode
Compiling .pio\build\esp01_1m\libbb5\ezTime_ID5751\ezTime.cpp.o
Compiling .pio\build\esp01_1m\src\main.cpp.o
Generating LD script .pio\build\esp01_1m\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\esp01_1m\lib289\SPI\SPI.cpp.o
.pio\libdeps\esp01_1m\ezTime_ID5751\src\ezTime.cpp:10:22: fatal error: EEPROM.h: No such file or directory

****************************************************************
* Looking for EEPROM.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:EEPROM.h"
* Web  > https://platformio.org/lib/search?query=header:EEPROM.h
*
****************************************************************
`
followed later on by more error messages 
`
Compiling .pio\build\esp01_1m\FrameworkArduino\FS.cpp.o
*** [.pio\build\esp01_1m\libbb5\ezTime_ID5751\ezTime.cpp.o] Error 1
In file included from .pio\libdeps\esp01_1m\WiFi_ID870\src\utility\spi_drv.cpp:22:0:
.pio\libdeps\esp01_1m\WiFi_ID870\src/utility/spi_drv.h:41:13: warning: 'initialized' defined but not used [-Wunused-variable]
 static bool initialized = false;
             ^
.pio\libdeps\esp01_1m\WiFi_ID870\src\utility\wifi_drv.cpp: In static member function 'static uint8_t WiFiDrv::getEncTypeNetowrks(uint8_t)':
.pio\libdeps\esp01_1m\WiFi_ID870\src\utility\wifi_drv.cpp:451:10: warning: converting to non-pointer type 'uint8_t {aka unsigned char}' from NULL [-Wconversion-null]
   return NULL;
          ^
.pio\libdeps\esp01_1m\WiFi_ID870\src\utility\wifi_drv.cpp: In static member function 'static int32_t WiFiDrv::getRSSINetoworks(uint8_t)':
.pio\libdeps\esp01_1m\WiFi_ID870\src\utility\wifi_drv.cpp:476:10: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
   return NULL;
          ^
`

If I now edit the include file ezTime.h  (part of the library) and move the line #4
#ifdef __cplusplus

down to line # 30

or remove the #ifdef ... #endif for __cplusplus altogether and then build again, everything works out ok!   Note the dependency graph now contains EEPROM and ESP8266WiFi as needed fo ezTime.  Also, no additional error messages are shown:
`
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 2.3.2 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 2.20603.191216 (2.6.3), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.2.7
|   |-- <SPI> 1.0        
|-- <ezTime> 0.8.3       
|   |-- <EEPROM> 1.0     
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\esp01_1m\src\main.cpp.o
Linking .pio\build\esp01_1m\firmware.elf
Retrieving maximum program size .pio\build\esp01_1m\firmware.elf
Checking size .pio\build\esp01_1m\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [===       ]  33.6% (used 27540 bytes from 81920 bytes)
PROGRAM: [====      ]  36.9% (used 280972 bytes from 761840 bytes)
================================================== [SUCCESS] Took 9.21 seconds ==================================================
...

`

My platformio.ini

`
[platformio]
description = Test of ezTime library

[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
upload_speed = 921600
monitor_speed = 115200
lib_ldf_mode = deep+
lib_deps = 
   WiFi
   ezTime ; 5751 by Rop Gonggrijp

`

The main.cpp 
`
#include "Arduino.h"
#include <WiFi.h>
#include <ezTime.h>

Timezone myTZ;
char ssid[] = "MEIN_WLAN";
char pass[] = "IST_GEHEIM";

void setup() {

	Serial.begin(115200);
	while (!Serial) { ; }		// wait for Serial port to connect. Needed for native USB port only
	WiFi.begin(ssid, pass);
	
	waitForSync();

	Serial.println();
	Serial.println("UTC:             " + UTC.dateTime());

}

void loop() 
{
  myTZ.setLocation(F("de"));
	Serial.println(myTZ.dateTime());
	events();
  delay(5000);
}

`

What can/should I do now?

with best regards

Volker Bandke

merlinschumacher, 2020-04-29:
I have the same issue.

ivankravets, 2020-06-03:
Please forward to https://community.platformio.org/

closed 2020-06-03 by ivankravets

vbandke, 2020-06-03:
> 
> 
> Please forward to https://community.platformio.org/

And how would I do this?

With best regards

Volker Bandke

ivankravets, 2020-06-03:
Login to our forum with your Github accout and create a new topic