platformio/platformio-vscode-ide #1488: Debugging: Option to monitor only custom generic or peripheral registers

ivankravets, 2020-01-16:
Currently, PlatformIO updates all known registers on the each ON-STOP event which comes from VSCode. It takes a lot of time to update all values for all registers. Think about how to make these Register Views full configurable. Developer should personally select which registers to monitor/update/

fanghuaqi, 2020-01-19:
I would like to see this feature to be added in future, the periperal and core registers cost too much time to get the value, we can step without live update, just update by manually click on it.

ivankravets, 2020-01-19:
You can collapse register views and PlatformIO will not update values.

fanghuaqi, 2020-01-20:
> 
> 
> You can collapse register views and PlatformIO will not update values.

Hi,  I tried to collapse the register views, but it didn't work for me.
Here is my vscode version:
[image]
[image]
[image]
[image]