platformio/platformio-vscode-ide #1494: strange behavior in finding the library

ijash, 2020-01-21:
i am using linux mint. and i was trying to use dallas temperature library, then, so many errors during compiling.
```
> Executing task in folder simoli-demo-v2: platformio run <

Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 6.0.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DallasTemperature> 3.8.0
|   |-- <OneWire> 2.3.5
|   |   |-- <FreeRTOS701>
|-- <OneWire> 2.3.5
|   |-- <FreeRTOS701>
Building in release mode
Compiling .pio/build/genericSTM32F103C8/libdcd/OneWire_ID1/OneWire.cpp.o
In file included from /home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:144:0:
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp: In member function 'void OneWire::begin(uint8_t)':
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:231:52: error: 'digitalPinToPinName' was not declared in this scope
 #define PIN_TO_BITMASK(pin)             ((uint32_t)digitalPinToPinName(pin))
                                                    ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:150:12: note: in expansion of macro 'PIN_TO_BITMASK'
  bitmask = PIN_TO_BITMASK(pin);
            ^~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:231:52: note: suggested alternative: 'digitalPinToBitMask'
 #define PIN_TO_BITMASK(pin)             ((uint32_t)digitalPinToPinName(pin))
                                                    ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:150:12: note: in expansion of macro 'PIN_TO_BITMASK'
  bitmask = PIN_TO_BITMASK(pin);
            ^~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp: In member function 'uint8_t OneWire::reset()':
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:55: error: 'PinName' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                       ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:55: note: suggested alternative: 'rename'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                       ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:81: error: 'STM_MODE_INPUT' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                 ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:81: note: suggested alternative: 'DIRECT_MODE_INPUT'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                 ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:97: error: 'GPIO_NOPULL' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                                                                                 ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:172:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(reg, mask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:97: note: suggested alternative: 'GPIO_INPUT_PU'
[... 605 lines cut ...]
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:41: error: 'pin_function' was not declared in this scope
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                         ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
/home/ijash/.platformio/lib/OneWire_ID1/util/OneWire_direct_gpio.h:238:41: note: suggested alternative: '__section'
 #define DIRECT_MODE_INPUT(base, pin)    pin_function((PinName)pin, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0))
                                         ^
/home/ijash/.platformio/lib/OneWire_ID1/OneWire.cpp:317:2: note: in expansion of macro 'DIRECT_MODE_INPUT'
  DIRECT_MODE_INPUT(baseReg, bitmask);
  ^~~~~~~~~~~~~~~~~
*** [.pio/build/genericSTM32F103C8/libdcd/OneWire_ID1/OneWire.cpp.o] Error 1
================================================================================= [FAILED] Took 0.81 seconds =================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
one of the problem is:
[image]
[image]

so, the compiler doesn't look in default  library directory in `home` folder

ijash, 2020-01-21:
In new project. the .ini file is using `genericSTM32F103C8`
[image]
but when i change it to `bluepill_f103c8`
[image]
it's just works.
I think the `genericSTM32F103C8` should be deleted.

ivankravets, 2020-06-03:
Please forward to https://community.platformio.org/

closed 2020-06-03 by ivankravets