platformio/platformio-vscode-ide #1496: Installation Manager

SalvadorScalia, 2020-01-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.41.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.17763, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp-10076AM0wRI1oxh4v\virtualenv-16.7.7\virtualenv.py", line 2628, in <module>
    main()
  File "C:\.platformio\.cache\tmp-10076AM0wRI1oxh4v\virtualenv-16.7.7\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\.platformio\.cache\tmp-10076AM0wRI1oxh4v\virtualenv-16.7.7\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\.platformio\.cache\tmp-10076AM0wRI1oxh4v\virtualenv-16.7.7\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\Gea Tecnolog�a\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\Gea Tecnolog�a\\AppData\\Local\\Programs\\Python\\Python37-32'
New python executable in C:\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\Gea Tecnología\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```

ivankravets, 2020-06-03:
H,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

closed 2020-06-03 by ivankravets

SalvatoreScalia, 2020-07-01:
I owe you an apology I pressed the F5 key and it was:

in plataformIO-> PROJECT TASK-> BUILD

I'm ashamed sorry