platformio/platformio-vscode-ide #1520: PermissionError: [WinError 5] in shutil.py shutil.move(tmp_dir, pkg_dir)

knopserl, 2020-02-03:
Before you close this ticket or point me to another team, please be aware, that I started the issue at the project team at esurna who redirected me to platformio who redirected me to the vscode team who redirected me to the platformio extention team. So I hope I finally found the proper support team.

````
I have a very tricky problem with vscode and platformio building and embedded projects for the ESP8266 devices. I always get an PermissionError: [WinError 5] in shutil.py shutil.move(tmp_dir, pkg_dir).
Together with some expert from the espurna project and platformio, we have done a lot of investigations over weeks and tried to analyze and investigate this error and finally it seem to be a problem in conjunction with vscode only.
I'm using Windows 10 1909 x64 and latest vscode 1.41.1, latest Python 3.8.1, latest platformio, git etc. etc.

The error is reproducable on two different PC's (running fresh install of lal latest versions of everything mentioned and required). It also occurs with different users (also Admin) or starting with admin rights. It's also not a directory permission problem, as I tried it on different PC's with different user and having all permission full open for everybody.

The PermissionError: [WinError 5] in shutil.py shutil.move(tmp_dir, pkg_dir) sometimes return also a locked by another user error, but we also temporary tried to have wait commands etc.

The reason why we think it's a problem in conjunction with vscode is, that the platformio build command works flawless outside vscode with the CLI interface but fails within vscode.

here is some more excerpt from the build log:

> Executing task: C:\Users\internet\.platformio\penv\Scripts\platformio.exe run --environment blitzwolf-bwshpx-v23-ota <

Processing blitzwolf-bwshpx-v23-ota (board: esp01_1m; platform: espressif8266@1.5.0; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting espurna.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing eeprom_rotate
git version 2.24.0.windows.2
Cloning into 'D:\mcdev\vscode\espurna\code\.pio\libdeps\blitzwolf-bwshpx-v23-ota\_tmp_installing-7v89pu0h-package'...
PermissionError: [WinError 5] Zugriff verweigert: 'D:\\mcdev\\vscode\\espurna\\code\\.pio\\libdeps\\blitzwolf-bwshpx-v23-ota\\_tmp_installing-7v89pu0h-package\\.git\\objects\\00\\b9d6c9a4a765a96958ffe1aaf5d95357c2aebc':
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\internet\.platformio\platforms\espressif8266@1.5.0\builder\main.py", line 375:
    target_elf = env.BuildProgram()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\internet\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\internet\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "c:\users\internet\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 793:
    rmtree(src)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 731:
    return _rmtree_unsafe(path, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 604:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 609:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Program Files (x86)\Python38-32\lib\shutil.py", line 607:
    os.unlink(fullname) 
````

ivankravets, 2020-06-03:
Fixed in the latest PIO Core. Please upgrade via `pio upgrade`.

closed 2020-06-03 by ivankravets

knopserl, 2020-06-08:
@ivankravets it really seem to work!!! Thanks a lot!!