platformio/platformio-vscode-ide #1522: spiffs upload error 148 (Esp-Wroom-2)

SurionAndrew, 2020-02-04:
Good day

I am using Visual Studio code with Platform IO (home 3.0.1) Core (4.1.0)
Plaforms: Espressif 8266 (2.3.2)

I seem to have trouble with the spiffs on the esp-wroom-2
**esp-01**
my project works perfectly on an esp-01,
platformio.ini contents:
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino

the files in my data folder upload correctly and my project behaves perfectly.

however when i try and change to the esp-wroom-2 my file upload failes

**ESP-Wroom-2**
for the ESP-Wroom-2 i change my platformio.ini file as per below:
'[env:esp_wroom_02]
; platform = https://github.com/platformio/platform-espressif8266.git
platform = espressif8266
board = esp_wroom_02
framework = arduino'

The firmware compile and uploads correctly, however uploading the spiffs results in the following console log:

Processing esp_wroom_02 (platform: espressif8266; board: esp_wroom_02; framework: arduino)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp_wroom_02.html
PLATFORM: Espressif 8266 2.3.2 > ESP-WROOM-02
HARDWARE: ESP8266 80MHz, 80KB RAM, 2MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 2.20603.191216 (2.6.3), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0), tool-mkspiffs 1.200.0 (2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- 1.0
| |-- 1.0
|-- 1.0
|-- 1.0
|-- 1.2
| |-- 1.0
|-- 1.0
| |-- 1.0
Building in release mode
Building SPIFFS image from 'data' directory to .pio\build\esp_wroom_02\spiffs.bin
*** [.pio\build\esp_wroom_02\spiffs.bin] Error 148
====================================================================================== [FAILED] Took 1.30 seconds ======================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

regards

Andrew

ivankravets, 2020-06-03:
Please forward to https://community.platformio.org/

closed 2020-06-03 by ivankravets

Icyrockton, 2021-05-01:
encounter the same problem , have you found a solution?