platformio/platformio-vscode-ide #1523: Installation Manager

cybersmurf, 2020-02-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.41.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.19559, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\sramek.AGER\.platformio\.cache\tmp-16364So0xfDe5yVEI\virtualenv-16.7.7\virtualenv.py", line 2628, in <module>
    main()
  File "C:\Users\sramek.AGER\.platformio\.cache\tmp-16364So0xfDe5yVEI\virtualenv-16.7.7\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\sramek.AGER\.platformio\.cache\tmp-16364So0xfDe5yVEI\virtualenv-16.7.7\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\sramek.AGER\.platformio\.cache\tmp-16364So0xfDe5yVEI\virtualenv-16.7.7\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Python27\lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\sramek.AGER\\.platformio\\penv\\Scripts\\python.exe'

New python executable in C:\Users\sramek.AGER\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\sramek.AGER\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```

ivankravets, 2020-03-01:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/1570#issuecomment-593128668? Does it work for you?

cybersmurf, 2020-03-23:
hello. i am very sorry. i missed your email, but new version works fine. please close this issue

closed 2020-03-23 by cybersmurf