platformio/platformio-vscode-ide #1524: Using build_flags include causes error #include errors detected

dmaixner, 2020-02-06:
I have prepared simplified project here:
https://github.com/dmaixner/IncludeError

Here is my c_cpp_properties file:
[c_cpp_properties.zip](https://github.com/platformio/platformio-vscode-ide/files/4165057/c_cpp_properties.zip)

There are **two problems in VSCode**:
1) #include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (D:\repos\IncludeError\src\main.cpp).
2) cannot open source file "stddef.h" (dependency of "Arduino.h")

But still, **project can be compiled without any problems.**

Steps, how I achieved
1) platformio updated in vscode, with all libs and platforms, etc.
2) created new project
3) created simple header file
4) updated platformio.ini with
```
build_flags = 
    -include "src/MyLib.h"
```

dmaixner, 2020-02-06:
What I have tried already:
https://github.com/platformio/platformio-vscode-ide/issues/1392
https://community.platformio.org/t/include-errors-detected/5858
https://community.platformio.org/t/vscode-platformio-arduino-and-includepath/10755

Maybe the problem is in C++ extension? https://github.com/microsoft/vscode-cpptools/issues/4928

closed 2020-09-18 by stale[bot]

mday64, 2020-11-14:
I had the same error when creating a new Arduino project for STM32 (as created by the new project wizard).  I used the C/C++: Reset IntelliSense Database and PlatformIO: Rebuild IntelliSense Index commands, and the error remained.  I built the project (via PlatformIO's Build task).  It built fine, but the error was still there. I quit VSCode, relaunched VSCode, and the error was gone.  I wonder if resetting/rebuilding the IntelliSense database was sufficient, but the UI needed to go back and check the reset/updated database (that is, maybe the UI didn't refresh after fixing the IntelliSense database)?