platformio/platformio-vscode-ide #1530: VS Code reports an extension issue

al-davis, 2020-02-10:
I've been using VScode for a couple of weeks without issue.  Suddenly odd behavior.  Uninstalled and downloaded 1.42 today Feb 8,2020.  Odd behavior continues.

Open new project with platformio 
add line printf("Hello World"); in loop()
click checkmark in bottom of page blue highlited area
popup at top of window "Select task to run" with build as only choice
click "build" & get the following in the terminal window.

> Executing task: msbuild /property:GenerateFullPaths=true /t:build /consoleloggerparameters:NoSummary <

msbuild : The term 'msbuild' is not recognized as the name of a cmdlet, function, script file, or operable 
program. Check the spelling of the name, or if a path was included, verify that the path is correct and try 
again.
At line:1 char:1
+ msbuild /property:GenerateFullPaths=true /t:build /consoleloggerparam ...
+ ~~~~~~~
    + CategoryInfo          : ObjectNotFound: (msbuild:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
_________________________________________________________________

Can't proceed, please help.

VS Code version: Code 1.42.0 (ae08d5460b5a45169385ff3fd44208f431992451, 2020-02-06T10:51:34.058Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i3-4170 CPU @ 3.70GHz (4 x 3691)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|3.91GB (1.14GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details><details><summary>Extensions (2)</summary>

Extension|Author (truncated)|Version
---|---|---
cpptools|ms-|0.26.3
platformio-ide|pla|1.10.0

ivankravets, 2020-02-13:
https://community.platformio.org/t/error-the-platformio-task-detection-didnt-contribute-a-task/8272

al-davis, 2020-02-13:
Tried this solution, it made no difference to MY problem.

al-davis, 2020-02-13:
I tried the proposed solution, it made no difference to my problem. 
Please re-open the issue, I still can't use VSCode / PlatformIO IDE.

ivankravets, 2020-02-13:
Could you open PlatformIO Project and make a screenshot of Left Sidebar > PlatformIO? Do you see any project tasks?

al-davis, 2020-02-14:
[image]
[image]

al-davis, 2020-02-14:
[image]

ivankravets, 2020-02-14:
Please provide an output from VSCode > Help > Toggle developer tools > Console

al-davis, 2020-02-14:
[image]

ivankravets, 2020-02-14:
Try to remove `%HOMEDIR%/.platformio/penv` folder and restart VSCode.

al-davis, 2020-02-14:
Renamed folder penv and restarted VSCode results in message in platformIO bottom blue line "PlatformIO: Installing PlatformIO IDE..."  with endless twirly and no internet activity on Win10 monitor except for a quick burst of activity on startup. 2  identical tries.

ivankravets, 2020-02-14:
Do you have antivirus tools in the system? Please check again errors in debug console.

al-davis, 2020-02-14:
[image]

Only Win10 integrated virus tools.  Just ran a quick scan, no issues.

ivankravets, 2020-02-14:
Please message me directly at https://community.platformio.org/u/ivankravets/

al-davis, 2020-02-14:
Is there a way to message you there without signing up ?  I signed up at GitHub last week and suddenly I get 50+ spam emails a day.  Can't we continue here?

ivankravets, 2020-02-15:
Please email `me` at `ikravets.com`

al-davis, 2020-02-15:
Done

al-davis, 2020-02-16:
Offer for remote desktop debugging respectfully declined.

ivankravets, 2020-02-16:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

al-davis, 2020-02-16:
C:\Users\User>echo %path%
C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\lenovo\easyplussdk\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\PuTTY\;C:\Program Files (x86)\AOMEI Backupper;C:\Users\User\AppData\Local\Microsoft\WindowsApps;C:\Users\User\AppData\Local\Programs\Microsoft VS Code\bin

C:\Users\User>

al-davis, 2020-02-16:
and I don't know how to get the others, I'll need guidance.

ivankravets, 2020-02-16:
What are your VSCode settings? Maybe you broke PlatformIO IDE extension settings? Did you turn off built-in PIO Core? VSCode > Preferences > and share JSON settings here. See 
[image]

al-davis, 2020-02-16:
"configurations": [
        {
            "name": "!!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags"
        },
        {
            "name": "Win32",
            "includePath": [
                "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/include",
                "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/src",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/cores/arduino",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/variants/standard",
                "C:/Users/User/.platformio/lib/XBee_ID6",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/AltSoftSerial-master",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/EEPROM/src",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/HID/src",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/NewSoftSerial-master",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/SPI/src",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/SoftwareSerial/src",
                "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/Wire/src",
                "C:/Users/User/.platformio/packages/tool-unity",
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "path": [
                    "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/include",
                    "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/src",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/cores/arduino",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/variants/standard",
                    "C:/Users/User/.platformio/lib/XBee_ID6",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/AltSoftSerial-master",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/EEPROM/src",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/HID/src",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/NewSoftSerial-master",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/SPI/src",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/SoftwareSerial/src",
                    "C:/Users/User/.platformio/packages/framework-arduino-avr/libraries/Wire/src",
                    "C:/Users/User/.platformio/packages/tool-unity",
                    ""
                ]
            },
            "defines": [
                "PLATFORMIO=40100",
                "ARDUINO_AVR_UNO",
                "F_CPU=16000000L",
                "ARDUINO_ARCH_AVR",
                "ARDUINO=10808",
                "__AVR_ATmega328P__",
                ""
            ],
            "intelliSenseMode": "clang-x64",
            "cStandard": "c11",
            "cppStandard": "c++11",
            "compilerPath": "\"C:/Users/User/.platformio/packages/toolchain-atmelavr/bin/avr-gcc.exe\" -mmcu=atmega328p"
        }
    ],
    "version": 4
}

{
	// See http://go.microsoft.com/fwlink/?LinkId=827846
	// for the documentation about the extensions.json format
	"recommendations": [
		"platformio.platformio-ide"
	]
}

/ AUTOMATICALLY GENERATED FILE. PLEASE DO NOT MODIFY IT MANUALLY

// PIO Unified Debugger
//
// Documentation: https://docs.platformio.org/page/plus/debugging.html
// Configuration: https://docs.platformio.org/page/projectconf/section_env_debug.html

{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            "executable": "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/.pio/build/uno/firmware.elf",
            "toolchainBinDir": "C:/Users/User/.platformio/packages/toolchain-atmelavr/bin",
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            },
            "internalConsoleOptions": "openOnSessionStart"
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (skip Pre-Debug)",
            "executable": "c:/Users/User/Documents/PlatformIO/Projects/HelloWorld/.pio/build/uno/firmware.elf",
            "toolchainBinDir": "C:/Users/User/.platformio/packages/toolchain-atmelavr/bin",
            "internalConsoleOptions": "openOnSessionStart"
        }
    ]
}

{
   // See https://go.microsoft.com/fwlink/?LinkId=733558
   // for the documentation about the tasks.json format
   "version": "2.0.0",
   "tasks": [
      {
         "label": "build",
         "command": "dotnet",
         "type": "shell",
         "args": [
            "build",
            "/property:GenerateFullPaths=true",
            "/consoleloggerparameters:NoSummary"
         ],
         "group": {
            "kind": "build",
            "isDefault": true
         },
         "presentation": {
            "reveal": "silent"
         },
         "problemMatcher": "$msCompile"
      }
   ]
}

al-davis, 2020-02-16:
[image]

This current page repeats endlessly thru VScode restarts and 'Reload Now's

ivankravets, 2020-02-16:
File -> Preferences -> Settings -> and click on icon which I draw above. Share JSON settings.

al-davis, 2020-02-16:
"workbench.startupEditor": "none",
    "C_Cpp.updateChannel": "Insiders",
    "editor.renderControlCharacters": true,
    "editor.tabSize": 3,
    "editor.autoClosingBrackets": "never",
    "editor.autoClosingOvertype": "never",
    "editor.autoClosingQuotes": "never",
    "editor.accessibilitySupport": "off",
    "editor.formatOnType": true,
    "platformio-ide.useBuiltinPIOCore": false,
    "task.autoDetect": "off",
    "workbench.enableExperiments": false,
    "workbench.list.automaticKeyboardNavigation": false,
    "workbench.view.alwaysShowHeaderActions": true,
    "extensions.autoUpdate": false,
    "update.enableWindowsBackgroundUpdates": false,
    "update.mode": "none",
    "telemetry.enableCrashReporter": false,
    "breadcrumbs.enabled": true,
    "editor.renderWhitespace": "all",
    "terminal.integrated.shell.windows": "C:\\WINDOWS\\System32\\cmd.exe"
}

ivankravets, 2020-02-16:
"platformio-ide.useBuiltinPIOCore": false,

ivankravets, 2020-02-16:
How should it work if you disabled it?

closed 2020-02-16 by ivankravets

al-davis, 2020-02-16:
I don't know that I did disable it.  I've sure never altered that file.

ivankravets, 2020-02-16:
This is a file for modified settings by users. if you didn't do this via UI or directly in the file, you might have a security problem with PC.

Please forward other questions to https://community.platformio.org/

al-davis, 2020-02-16:
I've modified the setting and the issue has not changed.