platformio/platformio-vscode-ide #1534: Installation Manager

dx168b, 2020-02-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.42.0
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.17134, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\reception\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```
  

_And have a problem witch building project for ESP8266 and Arduino framework afler last PIO core update:_
**"sdkconfig.h": No such file or directory**

ivankravets, 2020-02-13:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

dx168b, 2020-02-13:
```
C:\Users\reception>echo %PATH%
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Users\reception\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\reception\AppData\Local\Programs\Python\Python37\;C:\Users\reception\AppData\Local\Microsoft\WindowsApps;C:\Users\reception\AppData\Local\Programs\Microsoft VS Code\bin;

C:\Users\reception>python --version
Python 3.7.5

C:\Users\reception>pip --version
pip 19.2.3 from c:\users\reception\appdata\local\programs\python\python37\lib\site-packages\pip (python 3.7)

C:\Users\reception>pip list
Package    Version
---------- -------
pip        19.2.3
setuptools 41.2.0
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

I reinstalled the Platformio and cleared all the remnants of the previous installation. This problem was resolved, but the problem remained with the "sdkconfig.h" file. I get an error when compiling the project: 

`In file included from .pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.cpp:24:0:
.pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.h:26:23: fatal error: sdkconfig.h: No such file or directory`

My "platformio.ini" contains the following:

`[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
upload_speed = 921600
monitor_speed = 115200
lib_deps = 
	PubSubClient
	ArduinoJson
	U8g2
	DHT
	Hash
	ESP Async WebServer
	Adafruit NeoPixel
	Adafruit Unified Sensor
	Adafruit BME280 Library

[platformio]
description = Digital home sensor with display and night light`

ivankravets, 2020-02-13:
> I reinstalled the Platformio and cleared all the remnants 

Did you change Python?

> but the problem remained with the "sdkconfig.h" file.

https://github.com/xoseperez/espurna/pull/2146#issuecomment-585722749

dx168b, 2020-02-13:
Not. It is installed automatically.

closed 2020-02-13 by ivankravets

jclavier44, 2021-10-06:
I had same issue. I solved it by installing python 3.10 before PlatformIO installation and at the end of Pytohn installation I did "Disable path lentgh limit".