platformio/platformio-vscode-ide #1535: Installation Manager

laughingboyjj, 2020-02-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.42.0
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.16299, x64

  %23 Exception
  ```
  Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 397, in _error_catcher
    yield
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 479, in read
    data = self._fp.read(amt)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 62, in read
    data = self.__fp.read(amt)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\http\client.py", line 457, in read
    n = self.readinto(b)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\http\client.py", line 501, in readinto
    n = self.fp.readinto(b)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\ssl.py", line 1071, in recv_into
    return self.read(nbytes, buffer)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\ssl.py", line 929, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\cli\base_command.py", line 188, in main
    status = self.run(options, args)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\commands\install.py", line 345, in run
    resolver.resolve(requirement_set)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 196, in resolve
    self._resolve_one(requirement_set, req)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 359, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 307, in _get_abstract_dist_for
    self.require_hashes
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 199, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 1064, in unpack_url
    progress_bar=progress_bar
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 924, in unpack_http_url
    progress_bar)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 1152, in _download_http_url
    _download_url(resp, link, content_file, hashes, progress_bar)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 861, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\utils\hashes.py", line 75, in check_against_chunks
    for chunk in chunks:
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 829, in written_chunks
    for chunk in chunks:
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\utils\ui.py", line 156, in iter
    for x in it:
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 818, in resp_read
    decode_content=False):
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 531, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 496, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "C:\Users\Administrator.SC-201905242032\.platformio\python37\lib\contextlib.py", line 130, in __exit__
    self.gen.throw(type, value, traceback)
  File "C:\Users\Administrator.SC-201905242032\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 402, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.

	at c:\Users\Administrator.SC-201905242032\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:35084
	at ChildProcess.d (c:\Users\Administrator.SC-201905242032\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:200:13)
	at ChildProcess.cp.emit (c:\Users\Administrator.SC-201905242032\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-03-01:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/1570#issuecomment-593128668? Does it work for you?

laughingboyjj, 2020-03-02:
Thank you, I have unlocked! Python version cannot use 3.7, only 2.7.14

closed 2020-03-02 by laughingboyjj

ivankravets, 2020-03-02:
How did you resolve this issue?