platformio/platformio-vscode-ide #1537: ValueError: path is on mount 'C:', start on mount 'd:

olonsoft, 2020-02-14:
When I click the Clean button (PlatformIO: Clean) I get the following error:
```
ValueError: path is on mount 'C:', start on mount 'd:':
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 133: 
    env.PioClean(env.subst("$BUILD_DIR"))
  File "C:\Users\irakl\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 299:
    clean_rel_path = relpath(clean_dir)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\ntpath.py", line 562:
    path_drive, start_drive))
```
My project compiles and run normally. Only "Clean" command reports this error.

ivankravets, 2020-06-03:
Could you reproduce this issue again?

olonsoft, 2020-06-03:
Yes, I 've just tried:
```
Executing task: C:\Users\irakl\.platformio\penv\Scripts\platformio.exe run --target clean <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------- 
ValueError: path is on drive C:, start on drive D::
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 145:
    env.PioClean(env.subst("$BUILD_DIR"))
  File "C:\Users\irakl\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224: 
    return self.method(*nargs, **kwargs)
  File "C:\users\irakl\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 332:
    clean_rel_path = relpath(clean_dir)
  File "c:\users\irakl\.platformio\penv\lib\ntpath.py", line 529:
    % (path_prefix, start_prefix))
========================================================== [FAILED] Took 2.81 seconds ==========================================================
```

ivankravets, 2020-06-03:
What is your `platformio.ini`? Do you have a custom path for `build_dir`?