platformio/platformio-vscode-ide #1540: New version breaking few libraries

mvadu, 2020-02-16:
I was working on a project using [AsyncTCP ](https://github.com/me-no-dev/AsyncTCP)and [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) libraries. which was compiling fine. After a `platformio upgrade` I started getting bunch of errors (e.g. `'panic' was not declared in this scope`, `'ip_addr_t {aka struct ip_addr}' has no member named 'addr'` etc). Searching for solution led me to [this issue](https://github.com/me-no-dev/AsyncTCP/issues/81) and as suggested adding `lib_compat_mode=strict` solved the issue. Adding it here incase other folks are facing similar issues with other libraries.

orestv, 2020-02-16:
(I had a similar problem. To make this page more searchable by Google for other related problems:)

The `lib_compat_mode=strict` addition to `platformio.ini` is also a **confirmed workaround** for the following error when trying to use AsyncTCP:

`In file included from .pio/libdeps/remote/AsyncTCP_ID1826/src/AsyncTCP.cpp:24:0:
.pio/libdeps/remote/AsyncTCP_ID1826/src/AsyncTCP.h:26:23: fatal error: sdkconfig.h: No such file or directory
`

atanisoft, 2020-02-16:
this should be sent over to https://github.com/platformio/platformio-core as that is where the issue with LDF exists.

mvadu, 2020-02-16:
I think there were few issues reported in core like [3381](#https://github.com/platformio/platformio-core/issues/3381), but the actual description mentioned in there is little different than th eissues we got. 

As per that issue they have a fix ready [5cc9a32](https://github.com/platformio/platformio-core/commit/5cc9a328ab4d64c42f3cda714f65e150eca31128)

closed 2020-02-16 by mvadu