platformio/platformio-vscode-ide #1555: Installation Manager

Guihudge, 2020-02-24:
%23 Description of problem
  try to install, and python install at (C:\Users\guihudge\AppData\Local\Programs\Python\Python38\python.exe)

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.42.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\guihudge\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```

ivankravets, 2020-03-01:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/1570#issuecomment-593128668? Does it work for you?

Guihudge, 2020-03-02:
I have just, reinstall python and it work

closed 2020-03-02 by ivankravets