platformio/platformio-vscode-ide #1563: PIO Home loading too slow

fanghuaqi, 2020-02-26:
Hi, I am using PIO home in VSCode, I tried to click the PIO Home, but if the network is bad, it is always show loading, is it possible to have a choice that if the network to pio is bad, then let user choose work offline or have better experince of offline development.

I am using PIO in China, maybe the network to your server is not that good in each location of the world, so I think offline development in VSCode is important for us.

Thanks
Huaqi

ivankravets, 2020-06-03:
It should work fast. We cache recent news. Could you reproduce this issue with the latest PIO Core?

fanghuaqi, 2020-06-05:
> 
> 
> It should work fast. We cache recent news. Could you reproduce this issue with the latest PIO Core?

It still works slow when first loading, but when it loads, and click it again,  it is faster than before.

fanghuaqi, 2020-06-05:
This is a screenshot of my pio home.

[image]

ivankravets, 2020-06-05:
Thanks, fixed in https://github.com/platformio/platformio-core/commit/e0e97a36297852128016b2ba8360d77f64b276db

Please open PlatformIO IDE Terminal and type `pio update`. Restart VScode, you should not see broken images.

closed 2020-06-05 by ivankravets