platformio/platformio-vscode-ide #1574: Error when using debug:  .pioinit:16: Error in sourced command file:

jiladahe1997, 2020-03-03:
Here is my **_platformio.ini:_**
```
[env:genericSTM32F103RC]
platform = ststm32
board = genericSTM32F103RC
framework = stm32cube
build_flags = 
  -DF1  
  -IC:/.platformio/packages/framework-stm32cube/f1/Drivers/STM32F1xx_HAL_Driver/Src
upload_protocol = jlink
debug_tool = jlink
```

I'm using j-link and could compile & upload firmware correctly. But when I use debug, it shows the error as below:
[image]

Since the error is unreadable, it hard for me to find what happended. And after hours trying, I used the CLI: 
```
platformio debug --interface=gdb -x .pioinit
```
And it shows the real problem:
[image]

At last I fix the problem with this: [https://github.com/platformio/platformio-core/issues/1959](https://github.com/platformio/platformio-core/issues/1959)

<br />
<br />

**In summary, I have two suggesstions:** 
- Why the error is unreadable ? I guess it's a problem about encoding, and fix it would be great.
- Why the debug console don't output the error trace as the CLI does? output the error trace may be helpful for further more debug probloms in the furture.

_By the way, I want to make some contribution on this , but I could not find the source code of  the package:[platformio-vscode-debug](https://www.npmjs.com/package/platformio-vscode-debug), maybe its not public?_

Thanks for your reading.

ivankravets, 2020-06-03:
Please forward to https://community.platformio.org/

closed 2020-06-03 by ivankravets

marianky, 2020-08-07:
I am having the same exact error. I don't see how you were able to fix that problem from that other thread: platformio/platformio-core#1959 ?? 

Please advice I really need to be able to debug my code.

jiladahe1997, 2020-08-08:
> I am having the same exact error. I don't see how you were able to fix that problem from that other thread: [platformio/platformio-core#1959](https://github.com/platformio/platformio-core/issues/1959) ??
> 
> Please advice I really need to be able to debug my code.

Please try this:
```
pio upgrade --dev
pio update
```

ivankravets, 2020-08-08:
@marianky it means that there is an issue with debugging configuration if this errors raises. Could you share your `platformio.ini`?