platformio/platformio-vscode-ide #1577: Cannot Deal Dir/Files When Running in WSL

Philosophistoria, 2020-03-04:
Recently I have tried to make a platformio project with the [VSCode running on WSL](
https://code.visualstudio.com/docs/remote/wsl), 
in which mode, VSCode is just an interface to VSCode Server which running WSL and all procedures are run on WSL.

- I successfully installed platformio extension in the VSCode in WSL mode.
- But the platformio seems still behaving as the extension for WIN ver.
  -> the platformIO can neither read nor make directory and files. 
- Even so, the home directory is regarded as being in WSL. 
  -> for instance, platformIO cannot retrieve the directories structure and reflect it on the window of `New Project`. 
  - And if make a project, whose name is <projectname> in `home/project/` on WSL, it makes `home/"project\<projectname>"`
  - when seeing this project on WSL shell, it looks like a file "project\<projectname" exists under the directory "project/"
- P.S.: It seems the hyperlinks to the web pages from PIO HOM also doesn't work if on the WSL mode.

[image]

Is there any solution so far?

My environment is below
Windows 10 Pro 1909
WSL: Ubuntu 18.04.4 LTS
VSCode: 
```
Version: 1.42.1 (user setup)
Commit: c47d83b293181d9be64f27ff093689e8e7aed054
Date: 2020-02-11T14:45:59.656Z
Electron: 6.1.6
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.18363
```
PlatformIO: Core 4.2.1·Home 3.1.1

Thanks for your help, and if it's not suitable for here or already resolved, please lead me to the site.

ivankravets, 2020-06-03:
A temporary solution is to use VSCode > File > Open directly.

sean-mcmanus, 2020-06-29:
What does VSCode > File > Open mean? How does that workaround the issue?

pedvide, 2020-11-07:
I have the same issue, it seems platformio is trying to open files using the windows path separator `\` instead of the linux one. Is there any workaround this?

afroserom, 2021-04-20:
I am facing the same issue using WSL: Ubuntu -20.04 
The path of the created project end up being like this '/Documents/PlatformIO/Projects\one-plus' and it can't be read by PlatformIO

jynolen, 2021-04-22:
I think main issues comes from the usage of fs
```
static isPIOProjectSync(projectDir) {
    try {
      fs.accessSync(path.join(projectDir, 'platformio.ini'));
      return true;
    } catch (err) {}
    return false;
  }
```

```
pini = "/home/jynolen/D12/platformio.ini"
fs.accessSync(pini) <=> Exception
```

Rottenbeer, 2021-05-18:
I also noticed this when I tried to work with platformio yesterday. I'm using it on windows 10 to develop an arduino project in the WSL with Ubuntu 20.04.
I cannot open or create a project (Filesystem seperators are wrong, as @Philosophistoria already reported.
Opening the folder works and provides most of the functionality of the extension, but unfortunately you cannot use the library manager then (it requires a project to be open).
Is there a workaround to fix this? Do you know already if you can fix it soon?

Squirrelcoding, 2022-05-31:
I know this thread is a year old but has anybody come up with a good solution now?

unsplorer, 2022-06-18:
I did!. Its here : https://github.com/platformio/platformio-home/issues/2316#issuecomment-1151332626

Papierkorb, 2023-03-28:
@the-moog I fear that said PR is (sadly) closed simply because the maintainers reside in the Ukraine and may not have had the time to work on PlatformIO much due to a maniac trying to occupy their land.

I hope that they're fine .. and I also hope that this PR will get looked at because I also really need this.

ivankravets, 2023-03-28:
Sorry for the issue. FYI, we are working on the next-gen version of PIO Home and will use IDE's file explorer instead of virtual. 

A temporary solution is to manage project folders using the system file explorer and later just open them in the VSCode.

fludo, 2024-05-06:
Would't it be possible to add a "custom folder path" parameter to prevend unusable projet on creation with PIO Ide ?