platformio/platformio-vscode-ide #1581: Change the status bar position

gitterman, 2020-03-04:
I am on windows and have my taskbar at the bottom set to autohide to get more screen estate. 
I run platformio with VScode and trying to click something on the tiny platformIO status bar frequently has me raising the window taskbar which then overlays the platformIO status bar.
Thus, I would very much appreciate if I could relocate the platformio status bar to the top or to the side.
Alas I cannot find a setting for this.
Is it possible through some config file though?

ivankravets, 2020-06-03:
We can't :( VSCode API does not allow us to do this. https://github.com/Microsoft/vscode/issues?q=is%3Aissue+is%3Aopen+toolbar+sort%3Areactions-%2B1-desc

closed 2020-06-03 by ivankravets

n0t-Human, 2024-08-25:
But other extensions like I use an Android  extension it has it's custom buttons at the top. (One for compiling & other for sideloading the apk to the Android device connected over adb)