platformio/platformio-vscode-ide #1582: Library Manager - Install to Local Project Wrong Location

mriksman, 2020-03-05:
Hi,

First time user of PlatformIO. Off to a rocky start. Installed PlatformIO/VSCode and Python 3.8.2.

To get a feel for the software, I am looking at Library Manager and the lib_deps in Platform.ini.

Library Manager - Global; installs to `C:\Users\{username}\.platformio\lib\{libraryname}`
Using lib_deps in Platform.ini; installs to `{projectname}\.pio\libdeps\esp01_1m\{libraryname}`
Using the command line `platformio lib install {libraryname}` installs to `{projectname}\.pio\libdeps\esp01_1m\{libraryname}`

But the real confusing one is using
Library Manager - PROJECT LOCAL; installs to `C:\Users\{username}\.platformio\penv\Scripts\.pio\libdeps\esp01_1m\{libraryname}`

Which doesn't seem right...

pfeerick, 2020-03-06:
I can't confirm the latter - where the local / project library install option actually installs the option to, but I seen mention of, and experienced myself where it did not get installed into the `.pio\libdeps` of the project directory. So it's quite possible these behaviours are related.

pfeerick, 2020-03-06:
Related issue, which I posted against pio-home thinking it was more likely a fault in that module, rather than in the VSCode IDE extension... https://github.com/platformio/platformio-home/issues/374

ivankravets, 2020-03-06:
Duplicate of https://github.com/platformio/platformio-home/issues/374

closed 2020-03-06 by ivankravets