platformio/platformio-vscode-ide #1604: Could not load platform data

prying, 2020-03-15:
using the vscode extension when trying to download a platform (or try to remove one) i get the following error:
```
PIO Core Call Error : PlatformManager: Installing gd32v
Downloading...
Unpacking...
gd32v @ 1.1.2 has been successfully installed!

Error: Traceback (most recent call last):
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
 cli() # pylint: disable=no-value-for-parameter
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
 return self.main(*args, **kwargs)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
 rv = self.invoke(ctx)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
 return super(PlatformioCLI, self).invoke(ctx)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
 return _process_result(sub_ctx.command.invoke(sub_ctx))
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
 return _process_result(sub_ctx.command.invoke(sub_ctx))
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
 return ctx.invoke(self.callback, **ctx.params)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
 return callback(*args, **kwargs)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 325, in platform_install
 force=force,
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 103, in install
 force=force,
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 307, in install_packages
 self.pm.install(name, version, silent=silent, force=force)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 686, in install
 self.print_message(msg)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 608, in print_message
 click.echo("%s: %s" % (self.__class__.__name__, message), nl=nl)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\__main__.py", line 90, in <lambda>
 click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\__main__.py", line 87, in _safe_echo
 "%s\n" % (args[0] if args else "")
 File "c:\users\flynn\.platformio\penv\lib\site-packages\platformio\commands\home\rpc\handlers\piocore.py", line 55, in write
 value.decode() if is_bytes(value) else value
TypeError: 'unicode' does not have the buffer interface

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
 `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
 https://docs.platformio.org/page/faq.html

* Report this problem to the developers
 https://github.com/platformio/platformio-core/issues

============================================================
```

MHalter, 2020-05-18:
Hi,
similar for me, if I want to install the Adafruit IO libraries

```
Libraries: Could not install library
PIO Core Call Error: "Library Storage: C:\\Users\\Halter\\.platformio\\lib\
LibraryManager: Installing mid=714@ ^3.6.0
Adafruit IO Arduino @ 3.6.0 is already installed
Installing dependencies

Error: Traceback (most recent call last):
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 102, in main
 cli() # pylint: disable=no-value-for-parameter
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__
 return self.main(*args, **kwargs)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main
 rv = self.invoke(ctx)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke
 return super(PlatformioCLI, self).invoke(ctx)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke
 return _process_result(sub_ctx.command.invoke(sub_ctx))
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke
 return _process_result(sub_ctx.command.invoke(sub_ctx))
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke
 return ctx.invoke(self.callback, **ctx.params)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke
 return callback(*args, **kwargs)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func
 return f(get_current_context(), *args, **kwargs)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 155, in lib_install
 library, silent=silent, interactive=interactive, force=force
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 320, in install
 lib_id = self.search_lib_id(filters, silent, interactive)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 149, in search_lib_id
 % click.style(filters[\"name\"], fg=\"cyan\")
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in <lambda>
 click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 87, in _safe_echo
 \"%s\
\" % (args[0] if args else \"\")
 File \"c:\\users\\halter\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\home\\rpc\\handlers\\piocore.py\", line 55, in write
 value.decode() if is_bytes(value) else value
TypeError: 'unicode' does not have the buffer interface

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
 `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
 https://docs.platformio.org/page/faq.html

* Report this problem to the developers
 https://github.com/platformio/platformio-core/issues

===================
``

Does anyone have any suggestions for me?

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

closed 2020-06-03 by ivankravets

MHalter, 2020-06-04:
Hi,
thanks for the hint. I found a solution which works for me even without using the Beta. In my case an update from Python 2.7 to Python 3.8 fixes the problem.

Thanks
Mark