platformio/platformio-vscode-ide #1616: Installation Manager

Lemonhubs, 2020-03-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.43.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 425, in _error_catcher
    yield
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 507, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 62, in read
    data = self.__fp.read(amt)
  File "F:\python3.7\lib\http\client.py", line 447, in read
    n = self.readinto(b)
  File "F:\python3.7\lib\http\client.py", line 491, in readinto
    n = self.fp.readinto(b)
  File "F:\python3.7\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
  File "F:\python3.7\lib\ssl.py", line 1049, in recv_into
    return self.read(nbytes, buffer)
  File "F:\python3.7\lib\ssl.py", line 908, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\cli\base_command.py", line 186, in _main
    status = self.run(options, args)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\commands\install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 482, in prepare_linked_requirement
    hashes=hashes,
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 287, in unpack_url
    hashes=hashes,
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 159, in unpack_http_url
    link, downloader, temp_dir.path, hashes
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 303, in _download_http_url
    for chunk in download.chunks:
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\utils\ui.py", line 160, in iter
    for x in it:
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_internal\network\utils.py", line 39, in response_chunks
    decode_content=False,
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 564, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 529, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "F:\python3.7\lib\contextlib.py", line 130, in __exit__
    self.gen.throw(type, value, traceback)
  File "C:\Users\dele\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 430, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.

	at c:\Users\dele\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:35084
	at ChildProcess.d (c:\Users\dele\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\dele\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-03-20:
Could you try to manually install PlatformIO Core using installer Script? See documentation:
- https://docs.platformio.org/en/latest/core/installation.html#installer-script

Restart VSCode. Does it work for you?

darrentmorgan, 2020-05-17:
> Could you try to manually install PlatformIO Core using installer Script? See documentation:
> 
> * https://docs.platformio.org/en/latest/core/installation.html#installer-script
> 
> Restart VSCode. Does it work for you?

I've tried the above manual install, I get the following error: 
```
sickenxo@sickenxo-linux:~$ curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -o get-platformio.py
sickenxo@sickenxo-linux:~$ python3 get-platformio.py
Installer version: 0.3.0
Platform: Linux-5.4.0-29-generic-x86_64-with-glibc2.29
Python version: 3.7.6 (default, Apr 13 2020, 12:16:53) 
[GCC 9.2.1 20191008]
Python path: /home/sickenxo/.pyenv/versions/3.7.6/bin/python3
Creating a virtual environment at /home/sickenxo/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-4.3.3.tar.gz (187 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/sickenxo/.platformio/penv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bn98urq5/platformio/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bn98urq5/platformio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-spntinw5
         cwd: /tmp/pip-install-bn98urq5/platformio/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/sickenxo/.platformio/penv/lib/python3.7/site-packages/setuptools/__init__.py", line 19, in <module>
        from setuptools.dist import Distribution
      File "/home/sickenxo/.platformio/penv/lib/python3.7/site-packages/setuptools/dist.py", line 34, in <module>
        from setuptools import windows_support
      File "/home/sickenxo/.platformio/penv/lib/python3.7/site-packages/setuptools/windows_support.py", line 2, in <module>
        import ctypes
      File "/home/sickenxo/.pyenv/versions/3.7.6/lib/python3.7/ctypes/__init__.py", line 7, in <module>
        from _ctypes import Union, Structure, Array
    ImportError: libffi.so.6: cannot open shared object file: No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error: Could not install PlatformIO Core: Command '['/home/sickenxo/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
```

darrentmorgan, 2020-05-17:
I found this solution:
https://github.com/pypa/pipenv/issues/4194

It seems that when I have installed 20.04 Ubuntu it had corrupted the pyenv install of python 3.7.6. Re-installing this using:
```
pyenv install 3.7.6

```
and running the install script again resolved the issue for me. The main problem is the library libffi.so.6 missing from the pyenv venv.

Hope this helps.

closed 2020-06-03 by ivankravets