platformio/platformio-vscode-ide #1630: C/C++ IntelliSense not working properly

6Resh6, 2020-03-27:
Hi
I haven't paid attention, when the problem has started, may be week ago, or something like that...
When I try to rebuild IntelliSense index that's what happens:

Error: Processing nodemcu (platform: espressif8266; board: nodemcu; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcu.html
PLATFORM: Espressif 8266 2.4.0 > NodeMCU 0.9 (ESP-12 Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 64 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <MyArrayList_Class>
|-- <EEPROM> 1.0
|-- <MFRC522> 1.4.6
|   |-- <SPI> 1.0
|-- <MyArrayList_Variable>
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <FastLED> 3.3.3
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.4.0
|-- <TFT_ST7735> 1.0.1b1
|   |-- <SPI> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <MyParserClass>
|   |-- <MyArrayList_Class>
|   |-- <MyArrayList_Variable>
|-- <PubSubClient> 2.7
|-- <MyTimerClass>
|-- <SPI> 1.0
Building in release mode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 85: ordinal not in range(128):
  File "/Users/Roman/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 208:
    projenv.DumpIDEData()  # pylint: disable=undefined-variable
  File "/Users/Roman/.platformio/penv/lib/python2.7/site-packages/platformio/compat.py", line 78:
    obj, encoding=get_filesystem_encoding(), ensure_ascii=False, sort_keys=True
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 251:
    sort_keys=sort_keys, **kw).encode(obj)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py", line 210:
    return ''.join(chunks)
================================================================== [FAILED] Took 1.82 seconds ==================================================================

Could anybody help me to solve the issue?

Configurations:
macOS 10.15.3
Visual Studio Code v1.43.2
PIO Core 4.3.2a1 Home 3.1.1

Thanks.

ivankravets, 2020-06-03:
1. Please install the latest Python 3 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter
2. Remove `/Users/Roman/.platformio/penv/` folder
3. Restart VSCode.

closed 2020-06-03 by ivankravets

6Resh6, 2020-06-03:
Thank you very much for your respond, the problem already solved.