platformio/platformio-vscode-ide #1648: ESP32 build fails in linking firmware.elf

edouardreg, 2020-04-09:
Hello, I use VSCode + PlatformIO .
All was ok. But now I have just included ESPAsynchWebServer.h and as soon as I declare "AsyncWebServer server(80);" the build fails in Linking .pio\build\esp32dev\firmware.elf with multiple definitions errors.
------------------------------------------------------------------------------------------
plateformio.ini :
[env]
monitor_flags=
--raw
board_build.partitions = partitions_example.csv
[env:esp32dev]
;platform = espressif32
lib_compat_mode = strict
lib_ldf_mode = chain+
platform = espressif32@1.11.2
framework = arduino , espidf
board = esp32dev
monitor_speed = 115200
-----------------------------------------------------------------------------------
build logs :

```
Processing esp32dev (platform: espressif32@1.11.2; framework: arduino, espidf; board: esp32dev)
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
framework-arduinoespressif32 3.10004.200129 (1.0.4)
framework-espidf 3.30300.190916 (3.3.0)
tool-esptoolpy 1.20600.0 (2.6.0)
toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ strict
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <RTClib> 1.4.1
|   |-- <Wire> 1.0.1
|-- <Wire> 1.0.1
|-- <Adafruit Unified Sensor> 1.1.2
|   |-- <Adafruit ADXL343> 1.2.0
|   |   |-- <Wire> 1.0.1
|-- <Adafruit BME280 Library> 2.0.1
|   |-- <Adafruit Unified Sensor> 1.1.2
|   |   |-- <Adafruit ADXL343> 1.2.0
|   |   |   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <Adafruit MAX31865 library> 1.0.3
|   |-- <SPI> 1.0
|-- <NeoGPS> 4.2.9
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.1
Building in release mode
Linking .pio\build\esp32dev\firmware.elf
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_bad_exception()': /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc:58: multiple definition of std::__throw_bad_exception()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_bad_alloc()': functexcept.cc:(.text._ZSt17__throw_bad_allocv+0x0): multiple definition of std::__throw_bad_alloc()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
[... 13 lines cut ...]
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_out_of_range_fmt(char const*, ...)': functexcept.cc:(.text._ZSt24__throw_out_of_range_fmtPKcz+0x0): multiple definition of std::__throw_out_of_range_fmt(char const*, ...)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message_va+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_runtime_error(char const*)': functexcept.cc:(.text._ZSt21__throw_runtime_errorPKc+0x0): multiple definition of std::__throw_runtime_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_range_error(char const*)': functexcept.cc:(.text._ZSt19__throw_range_errorPKc+0x0): multiple definition of std::__throw_range_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_overflow_error(char const*)': functexcept.cc:(.text._ZSt22__throw_overflow_errorPKc+0x0): multiple definition of std::__throw_overflow_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_underflow_error(char const*)': functexcept.cc:(.text._ZSt23__throw_underflow_errorPKc+0x0): multiple definition of std::__throw_underflow_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_ios_failure(char const*)': functexcept.cc:(.text._ZSt19__throw_ios_failurePKc+0x0): multiple definition of std::__throw_ios_failure(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_system_error(int)': functexcept.cc:(.text._ZSt20__throw_system_errori+0x0): multiple definition of std::__throw_system_error(int)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_int+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_future_error(int)': functexcept.cc:(.text._ZSt20__throw_future_errori+0x0): multiple definition of std::__throw_future_error(int)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_int+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function std::__throw_bad_function_call()': functexcept.cc:(.text._ZSt25__throw_bad_function_callv+0x0): multiple definition of std::__throw_bad_function_call()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

I have tried update librairies, plateform..., ldf options : strict, chain+, no succes.
Thanks for your help

edouardreg, 2020-04-09:
I made some other test, it seems that ESP Asynch WebServer can't build in any "Arduino as component" projetct (framework : espidf , Arduino).
Here build logs from "simple web server" put in "Arduino as component" wifiscan exemple main.cpp : same error
```
> Executing task in folder testwifiscan: C:\Users\edoua\.platformio\penv\Scripts\platformio.exe run <

Processing esp32dev (platform: espressif32; framework: arduino, espidf; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - framework-espidf 3.30300.190916 (3.3.0)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\AsyncWebSocket.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\SPIFFSEditor.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\WebAuthentication.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\WebHandlers.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\WebRequest.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\WebResponses.cpp.o
Compiling .pio\build\esp32dev\liba5a\ESP Async WebServer_ID306\WebServer.cpp.o
Archiving .pio\build\esp32dev\liba5a\libESP Async WebServer_ID306.a
Linking .pio\build\esp32dev\firmware.elf
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_bad_exception()':
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc:58: multiple definition of `std::__throw_bad_exception()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_bad_alloc()':     
functexcept.cc:(.text._ZSt17__throw_bad_allocv+0x0): multiple definition of `std::__throw_bad_alloc()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_bad_cast()':      
functexcept.cc:(.text._ZSt16__throw_bad_castv+0x0): multiple definition of `std::__throw_bad_cast()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_bad_typeid()':    
functexcept.cc:(.text._ZSt18__throw_bad_typeidv+0x0): multiple definition of `std::__throw_bad_typeid()'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_logic_error(char const*)':
functexcept.cc:(.text._ZSt19__throw_logic_errorPKc+0x0): multiple definition of `std::__throw_logic_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_domain_error(char 
const*)':
functexcept.cc:(.text._ZSt20__throw_domain_errorPKc+0x0): multiple definition of `std::__throw_domain_error(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
c:/users/edoua/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(functexcept.o): In function `std::__throw_invalid_argument(char const*)':
functexcept.cc:(.text._ZSt24__throw_invalid_argumentPKc+0x0): multiple definition of `std::__throw_invalid_argument(char const*)'
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception_message+0x0): first defined here
[... cut ...]
.pio\build\esp32dev\libcxx.a(cxx_exception_stubs.cpp.o):cxx_exception_stubs.cpp:(.text.__cxx_fatal_exception+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
==================================================================================== [FAILED] Took 38.87 seconds ====================================================================================
The terminal process terminated with exit code: 1
```
platformio.ini : 

```[env:esp32dev]
platform = espressif32
framework = arduino, espidf
board = esp32dev
monitor_speed = 115200
```

ivankravets, 2020-06-03:
Please forward to https://community.platformio.org/

closed 2020-06-03 by ivankravets