platformio/platformio-vscode-ide #1650: Extension issue

WillOmae, 2020-04-10:
- Issue Type: `Bug`
- Extension Name: `platformio-ide`
- Extension Version: `1.10.0`
- OS Version: `Windows_NT x64 10.0.18363`
- VSCode version: `1.44.0`

```json 
{
	"messages": [],
	"activationTimes": {
		"codeLoadingTime": 14051,
		"activateCallTime": 32,
		"activateResolvedTime": 0,
		"activationReason": {
			"startup": true,
			"extensionId": {
				"value": "platformio.platformio-ide",
				"_lower": "platformio.platformio-ide"
			},
			"activationEvent": "*"
		}
	},
	"runtimeErrors": []
}
```

WillOmae, 2020-04-12:
The problem persists. This happens even when working on other projects that do not require the extension.

ivankravets, 2020-06-03:
Could you reproduce this issue again?

WillOmae, 2020-06-10:
I disabled the extension and now only enable it for the projects that need it. I haven't experienced the long startup time error since.

closed 2020-06-10 by WillOmae