platformio/platformio-vscode-ide #1681: memset() crashes debugger when size element is a variable on SAMD21

ASL07, 2020-04-23:
Hi
I am debugging a SAMD21 mcu (Arduino mkr)
Whenever the code reaches a memset() and the size arg is a variable, the debugger crashes with var-create error

This is really annoying, is there any workaround?

Please,  see below a crash on Arduino Core the RingBuffer class

[image]

ASL07, 2020-04-24:
I have tested this using Cortex debug in VSCode, and it works without any problems

reese-grimsley, 2020-12-11:
I believe I am seeing a similar issue myself with a custom D51 board. I am curious if anyone ever found a fix for this, but in my case, the issue is that memset (or memcp or snprintf or many other standard functions) will attempt to branch to some strange location that doesn't actually pertain to where that function would be expected to go. This usually generates a hardfault as it tries executing some nonsense bits as instructions. 

Unclear if this is from PIO or not, but that standard library is coming from somewhere, and compiler directives to _not_ link the standard library (-nostdlib) has no impact.