platformio/platformio-vscode-ide #1697: Feature request: Custom buttons and commands in VSC toolbar

chepo92, 2020-05-01:
I know I can make custom task and the build button task can be customized with one in settings
https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode-settings

But what about the others?
As example:
-  I want a button that Uploads code without building/checking and that doesn't do autoclean
- A button to show available COM's
- Etc

It would be nice to have easy to customize buttons in the toolbar.

pak-man, 2021-09-09:
I second this. Would be so so ergonomic / useful. Even without custom icon just a short label + tooltip on mouseover would be good enough.

closed 2022-12-18 by ivankravets

ivankravets, 2022-12-18:
Implemented in the latest development version (upcoming PlatformIO IDE 3.0). Please re-test with the pre-release version. See the instructions below:

[image]

-----

You will need to edit [platformio-ide.toolbar](https://github.com/platformio/platformio-vscode-ide/blob/develop/package.json#L799) option. See default declarations https://github.com/platformio/platformio-vscode-ide/blob/develop/package.json#L856. The `text` field can contain icons, see https://code.visualstudio.com/api/references/icons-in-labels#icon-listing

The full list of commands can be found via VSCode > Settings > Keybindings. You can declare multiple commands for one button.

We would be thankful for your feedback.

----

# Example

```json
    "platformio-ide.toolbar": [
        {
            "text": "Clean",
            "commands": [
                {
                    "id": "workbench.action.tasks.runTask",
                    "args": [
                        "PlatformIO: Clean"
                    ]
                }
            ]
        },
        {
            "text": "$(list-selection)",
            "commands": [
                {
                    "id": "platformio-ide.runPIOCoreCommand",
                    "args": "pio device list"
                }
            ]
        },
        {
            "text": "$(terminal)",
            "commands": [
                {
                    "id": "workbench.action.terminal.sendSequence",
                    "args": {
                        "text": "echo 1\n"
                    }
                }
            ]
        }
    ]
```

[image]

tookitogo, 2023-01-06:
> Implemented in the latest development version (upcoming PlatformIO IDE 3.0). Please re-test with the pre-release version. See the instructions below:
> You will need to edit [...]

OK, but how would one add "Upload and Monitor"? There is no command ID (e.g. `platformio-ide.uploadAndMonitor`) for that. (It's not obvious to me, despite poring over the code, where the command IDs themselves are defined.)

sivar2311, 2023-01-11:
> OK, but how would one add the aforementioned "Upload and Monitor"? There is no command ID (e.g. `platformio-ide.uploadAndMonitor`) for that. (It's not obvious to me, despite poring over the code, where the command IDs themselves are defined.)

I had the same idea in mind 😅
This is my solution:
```JSON
        {
            "text": "$(debug-alt)",
            "tooltip": "PlatformIO: Upload & Monitor",
            "commands": [
                {
                    "id": "workbench.action.tasks.runTask",
                    "args": [
                        "PlatformIO: Upload and Monitor"
                    ]
                }
            ]
        },
```

Hieromon, 2023-02-07:
@ivankravets Hello, 
Regarding associating the Upload and Monitor task to the toolbar, I think your intention is probably to assume [@sivar2311's solution](https://github.com/platformio/platformio-vscode-ide/issues/1697#issuecomment-1378364985). However, it does not correctly target the active environment when there are multiple environments.
Because it can not make the current active environment explicit to `pio run`.

**Actual command line passed to the vscode terminal** with the [above way](https://github.com/platformio/platformio-vscode-ide/issues/1697#issuecomment-1378364985).

It `--environment` argument is missing.

```
C:\Users\[CURRENT_USER]\.platformio\penv\Scripts\platformio.exe run --targert upload --target monitor
```

So, it seems to me that solving this problem is also related #2623 to making the Upload and Monitor task practical from the Toolbar even in the case of multiple environments.

Can I expect this to be resolved by fixing #2623 in a drastic way? Or have other workarounds?

Jason2866, 2023-02-19:
We encountered the same issue in Tasmota. https://github.com/arendst/Tasmota/discussions/17997

ivankravets, 2023-02-24:
We added support for the new "platformio-ide.uploadAndMonitor" command. You will be able to use 
```json
    "platformio-ide.toolbar": [
        {
            "text": "$(debug-alt)",
            "tooltip": "PlatformIO: Upload & Monitor",
            "commands": "platformio-ide.uploadAndMonitor"
        }
    ]
```

It will be a part of the PlatformIO IDE for VCCode 3.0.1.

Jason2866, 2023-02-24:
@ivankravets  Thx, do i understand correctly that the `--environment` is still not there?
This will not solve the issues in custom platformio toolbar.

ivankravets, 2023-02-24:
The `platformio-ide.uploadAndMonitor` command automatically handles the active environment. The issue was before with running bare `workbench.action.tasks.runTask` -> `PlatformIO: Upload and Monitor`

Hieromon, 2023-03-16:
Upload & Monitor with an active environment was achieved by the following settings:

```json
{
  "text": "$(arrow-right)",
  "tooltip": "PlatformIO: Upload & Monitor",
  "commands": [
    {
      "id": "workbench.action.tasks.runTask",
      "args": "PlatformIO: Upload and Monitor (${command:platformio-ide.activeEnvironment})"
    }
  ]
},
```

Thanks.

commit 1e449aac (2022-12-18): Configure custom buttons and commands in PlatformIO Toolbar with a new `platformio-ide.toolbar` configuration option // Resolve #1697