platformio/platformio-vscode-ide #1705: WHITE SPACE ISSUE

romocitto88, 2020-05-04:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN

https://github.com/platformio/platformio-vscode-ide/issues
%23 Configuration

VSCode: 1.44.2
PIO IDE: v1.10.0
System: Windows 10 home

%23 Exception

PIO Core Call Error: "The current working directory C:\\Users\\pippo will be used for the project.\r\n\r\n
The next files/directories have been created in C:\\Users\\pippo\r\n
include - Put project header files here\r\n
lib - Put here project specific (private) libraries\r\n
src - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\n
Error: Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)\r\n
--------------------------------------------------------------------------------\r\n
Verbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html\r\nPLATFORM: Espressif 32 1.12.1 > Espressif ESP32 Dev Module\r\nHARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash\r\nDEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)\r\nPACKAGES: \r\n 
- framework-espidf 3.40000.200303 (4.0.0) \r\n 
- tool-cmake 3.16.5 \r\n 
- tool-esptoolpy 1.20600.0 (2.6.0) \r\n 
- tool-idf 1.0.1 \r\n 
- tool-mconf 1.4060000.20190628 (406.0.0) \r\n 
- tool-ninja 1.9.0 \r\n 
- toolchain-esp32ulp 1.22851.190618 (2.28.51) \r\n 
- toolchain-xtensa32 2.80200.200226 (8.2.0)\r\n
Installing ESP-IDF's Python dependencies\r\n\
"c:\\users\\pluto" it is not recognized as an internal or external command, \ r \ n
an executable program or batch file.\r\n
*** Error 1\r\n
Error: Detected a whitespace character in project paths.\r\n
========================== [FAILED] Took 1.33 seconds =========================="

I'm tring to open a new project for esp32.
It seems the problem is a white space in the path after "pluto". How to solve it without change folder location?

ivankravets, 2020-06-03:
This is an issue of ESP-IDF. See details and workaround https://docs.platformio.org/en/latest/frameworks/espidf.html#limitations

closed 2020-06-03 by ivankravets

Nathan-ma, 2020-08-18:
> This is an issue of ESP-IDF. See details and workaround https://docs.platformio.org/en/latest/frameworks/espidf.html#limitations

Please, can you be a bit more specific? I've tried to add what the link has suggested and yet no luck

[image]

ivankravets, 2020-08-19:
> .... in a project path...

Does your full project path contain a space? This is an issue with ESP-IDF, not PlatformIO.