platformio/platformio-vscode-ide #1716: .vscode\c_cpp_properties.json has wrong name

mvadu, 2020-05-13:
with latest vscode I am seeing the auto generated `\.vscode\c_cpp_properties.json` file has incorrect name which causes the UI to show this 
[image]

```json
"configurations": [
        {
            "name": "!!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags"
        },
```

Manually editing this wont work since this file is generated by the platformio extn.

ivankravets, 2020-06-03:
Please open PlatformIO Project, add envs into `platformio.ini` and it will disappear.

mvadu, 2020-06-07:
I am sorry @ivankravets , this is what I have in `platformio.ini`, what additional envs i need to add?

```
[env:lolin32]
platform = espressif32
board = lolin32 
#esp32doit-devkit-v1
framework = arduino
#baud rate for platform io serial monitor
monitor_speed=115200
extra_scripts =
  pre:scripts/prebuild.py
```

ivankravets, 2020-06-07:
Open any C/C++ file and click on that `Warning` text on toolbar. Do you see other options, such Win?

mvadu, 2020-06-07:
I do, there is `Win32` as another option.
[image]

Also if I manually edit `.vscode\c_cpp_properties.json` and replace `name` in that json to say `PlatformIO` IDE picks it up immediately and shows it.. 

[image]

ivankravets, 2020-06-07:
Just select `Win32`

closed 2020-06-07 by ivankravets