platformio/platformio-vscode-ide #1721: Zephyr RTOS West support

Nukersson, 2020-05-16:
Can you add Zephyr RTOS west support? Or at least the possibility to define zephyr framework by locating zephyr's path on local machine?

valeros, 2020-06-03:
Hi @Nukersson! Using `west` with PlatformIO will create a lot of confusion mainly because Zephyr packages are manged solely by PlatformIO.

closed 2020-06-03 by valeros

camiloVargasTechnaid, 2024-08-09:
Hello!

Is there an option to use west with Platformio?, I'm working with Zephyr and for me, PlatformIO works really well, but in this moment I can't run MCUboot config in this because there are many errors associated with west.

Running example: https://github.com/mcu-tools/mcuboot/blob/main/samples/zephyr/hello-world/src/main.c