platformio/platformio-vscode-ide #1742: No boards loaded during initial PIO install

k5map, 2020-05-27:
I just installed VSCode and PIO today following the instructions.  When I started to create my first project, the board count was zero.  I read in a previous issue where the fix was to run PlatformIO update.  I did that but still there are no boards listed.

valeros, 2020-05-27:
Hi @k5map ! What do you mean by 
> the board count was zero

Are you able import an example and compile it?

k5map, 2020-05-28:
When I go to open a New Project or Import an Arduino Project, it shows I have 0 boards available (screenshot attached).
[image]

valeros, 2020-05-28:
Have you installed any platform? Try to run `pio platform install %platform_name%`, for example `pio platform install espressif8266` in the IDE terminal.

k5map, 2020-05-29:
I don't believe as I following the installation of PIO and that's it.  How do I get the IDE terminal to open?  When I enter those commands in the VS terminal window, I get a invalid command.

valeros, 2020-05-29:
You can install any platform using PlatformIO Home
[image]

closed 2020-06-03 by ivankravets