platformio/platformio-vscode-ide #1751: Built-in core uses Python 2.7

jamesmyatt, 2020-06-02:
According to the core installation instructions ( https://docs.platformio.org/en/latest/core/installation.html#), it's recommended to use Python 3+. However, I just re-installed the vscode-ide and the virtual (conda) environment it created uses Python 2.7. Is there a way to fix this?

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

closed 2020-06-03 by ivankravets

jamesmyatt, 2020-06-15:
Thanks. I'll try it later