platformio/platformio-vscode-ide #1755: Start PIO Home Server

223222981, 2020-06-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.45.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1 - Local (0.0.0.0:55347)
    at c:\Users\echo\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:201:70
    at processTicksAndRejections (internal/process/task_queues.js:85:5)
  ```

ivankravets, 2020-06-03:
Do you have enabled defender's firewall?

closed 2020-06-05 by 223222981

ivankravets, 2020-06-05:
How did you resolve this issue?

223222981, 2020-06-06:
> How did you resolve this issue?

Thank you for your help. I have solved the problem by trying here