platformio/platformio-vscode-ide #1758: Start PIO Home Server

iu1jvo, 2020-06-04:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.45.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.17763, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\giuliano.favro\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\giuliano.favro\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 33, in <module>
    import twisted
ImportError: No module named twisted

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\giuliano.favro\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\giuliano.favro\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\giuliano.favro\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-05:
Do you use Antivirus tool?

ivankravets, 2020-06-15:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1787

closed 2020-06-15 by ivankravets