platformio/platformio-vscode-ide #1762: Hot key bug

im-AMS, 2020-06-05:
Hi I have installed it in vscode.
If i have opened my python projects folder and arduino folder, and i run my python file with `ctrl+F5` platformio gets triggered for no reason. and if i open my arduino code and `ctrl+alt+b` i get 
`command 'platformio-ide.build' not found`
Its the same case for uploading with
`ctrl+alt+u`

ivankravets, 2020-06-05:
Did you enable VSCode to activate PlatformIO extension only on PlatformIO project in VScode settings?

im-AMS, 2020-06-05:
Ahhm... I'm new to vscode.. can u please tell me how do I check it..

ivankravets, 2020-06-05:
If you work on Python project, just disable PlatformIO this project. Go to Extensions View and click on ⚙️  and select "Disable for Workspace)

closed 2020-06-05 by ivankravets