platformio/platformio-vscode-ide #1767: Major problem with PlatformIO on Mac!

wangmings, 2020-06-06:
Major problem with PlatformIO on Mac!

Problem description: also like the upgrade of the library in PlatformIO caused the normal use of PlatformIO! Stuck in the loading interface! Vs code was tried to delete several times! Can't solve this problem!

The problem is shown in figure

Problem description:
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.36.0' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

    at /Users/mac/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:11965
    at ChildProcess.d (/Users/mac/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:8215)
    at ChildProcess.emit (events.js:203:13)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)

[image]

ivankravets, 2020-06-11:
Please click on PlatormIO Icon in the left sidebar > Misc > PlatformIO Core CLI and type in terminal
```
pio home
```

Does it work now?

ivankravets, 2020-06-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1777

closed 2020-06-11 by ivankravets