platformio/platformio-vscode-ide #1777: Start PIO Home Server

carlosdelfino, 2020-06-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.45.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.37.0' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\Admin\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\Admin\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\Admin\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-11:
Please click on PlatormIO Icon in the left sidebar > Misc > PlatformIO Core CLI and type in the terminal
```
pio home
```

Does it work now?

closed 2020-10-03 by stale[bot]

gaafouri, 2020-10-13:
@ivankravets  thanks 
i have this problem
Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'

at C:\Users\Ahmed\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:12126
at ChildProcess.s (C:\Users\Ahmed\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:4900)
at ChildProcess.emit (events.js:194:13)
at ChildProcess.cp.emit (C:\Users\Ahmed\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:998:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:265:5)

dj-ryan, 2020-10-13:
This is the same problem for me:

`Error: Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64' 
at c:\Users\golia\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:12033`
`at ChildProcess.s (c:\Users\golia\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807) at ChildProcess.emit (events.js:223:5)` 
`at ChildProcess.cp.emit (c:\Users\golia\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)` 
`at maybeClose (internal/child_process.js:1021:16) at Process.ChildProcess._handle.onexit (internal/child_process.js...`

gaafouri, 2020-10-13:
PS C:\Users\Ahmed\Documents\PlatformIO\Projects\arduino nano> pio home
********************************************************************************************************************************************************************************************************
Obsolete PIO Core v5.0.1 is used (previous was 5.0.2b2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
********************************************************************************************************************************************************************************************************
Tool Manager: Installing platformio/contrib-pysite @ ~2.39.0
Error: Could not find the package with 'platformio/contrib-pysite @ ~2.39.0' requirements for your system 'windows_amd64'
PS C:\Users\Ahmed\Documents\PlatformIO\Projects\arduino nano> pio update

gaafouri, 2020-10-14:
resolved change the version of python 3.9 to any other version

czilibili, 2023-04-07:
got same error while i tried to install platformio ide  
i`ve done 3 things
0) checked python version installed it is python 3.10.xx
1) installed python-is-python3 package
2) installed pip on my user account
3) and exported path export PATH="$HOME/.local/bin:$PATH" like this to run pip from diffreent locations