platformio/platformio-vscode-ide #1779: Start PIO Home Server

antiguaMob, 2020-06-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.45.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Python27\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Python27\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Python27\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Python27\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Python27\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Python27\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Python27\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\antigua\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 33, in <module>
    import twisted
ImportError: No module named twisted

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\antigua\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\antigua\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.EventEmitter.emit (domain.js:476:20)
	at ChildProcess.cp.emit (c:\Users\antigua\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1758

Could you reproduce this issue after VSCode restart?

antiguaMob, 2020-06-11:
It is repeated at each start, Python 2.7 and Python 3.8 are present in the system. In the console, the main one is 3.8. I tried to install beta 2.0, it does not start with me, it constantly tries to reinstall.
 
 
 
--
Алексей Агальцов

ivankravets, 2020-06-11:
Could you try PlatformIO IDE for VSCode 2.0 Beta? Does it work?
- https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097

antiguaMob, 2020-06-11:
It didn't work out until the end. There is an installation, something downloads from the network, asks to restart VS code . after restarting, the process is repeated indefinitely.
 
 
 
--
Алексей Агальцов

ivankravets, 2020-06-11:
Do you have antivirus tools in the system that block PlatformIO? Would you like to provide a remote session to debug this issue? Thanks! 
- https://anydesk.com/en

You can send me an ID to me @ ikravets.com

antiguaMob, 2020-06-12:
installing the beta version looks like this
 
 
 
--
Алексей Агальцов

ivankravets, 2020-06-13:
Try to remove “ C:\Users\antigua\.platformio\packages\contrib-pysite” folder and restart VSCode. Did it help?

HeartlandAgTech, 2020-08-31:
I figured out a similar issue. With the release of 1.10.0, the PIO core needed to be reinstalled for me.

The note in this issue helped resolve the problem: #1353

"Please remove HOME_DIR/.platformio/penv folder after update to 1.10.0 and restart VSCode."

ivankravets, 2020-08-31:
@HeartlandAgTech what is your OS?

HeartlandAgTech, 2020-08-31:
Windows 10.

closed 2020-10-07 by stale[bot]