platformio/platformio-vscode-ide #1785: Installation Manager

wavehf, 2020-06-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.46.0
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.17763, x64

  %23 Exception
  ```
  Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 425, in _error_catcher
    yield
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 507, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 62, in read
    data = self.__fp.read(amt)
  File "C:\Users\Think\.platformio\python37\lib\http\client.py", line 457, in read
    n = self.readinto(b)
  File "C:\Users\Think\.platformio\python37\lib\http\client.py", line 501, in readinto
    n = self.fp.readinto(b)
  File "C:\Users\Think\.platformio\python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
  File "C:\Users\Think\.platformio\python37\lib\ssl.py", line 1071, in recv_into
    return self.read(nbytes, buffer)
  File "C:\Users\Think\.platformio\python37\lib\ssl.py", line 929, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\cli\base_command.py", line 188, in _main
    status = self.run(options, args)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\cli\req_command.py", line 185, in wrapper
    return func(self, options, args)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\commands\install.py", line 333, in run
    reqs, check_supported_wheels=not options.target_dir
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 179, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 362, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 314, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 469, in prepare_linked_requirement
    hashes=hashes,
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 259, in unpack_url
    hashes=hashes,
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 130, in get_http_url
    link, downloader, temp_dir.path, hashes
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 281, in _download_http_url
    for chunk in download.chunks:
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\cli\progress_bars.py", line 166, in iter
    for x in it:
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_internal\network\utils.py", line 39, in response_chunks
    decode_content=False,
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 564, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 529, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "C:\Users\Think\.platformio\python37\lib\contextlib.py", line 130, in __exit__
    self.gen.throw(type, value, traceback)
  File "C:\Users\Think\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 430, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.

	at c:\Users\Think\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:35084
	at ChildProcess.d (c:\Users\Think\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\Think\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-11:
> HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.

Could you try to restart VSCode? There is an issue with an Internet connection.

wavehf, 2020-06-12:
I restarted VSCode several times, but nothing help, then i uninstalled Kendryte IDE, the problem fixed

closed 2020-06-12 by wavehf