platformio/platformio-vscode-ide #1787: Start PIO Home Server

davidpkgray, 2020-06-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.46.0
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "c:\users\david\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\david\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\david\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\david\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\david\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\david\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\david\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 33, in <module>
    import twisted
ImportError: No module named twisted

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\david\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\david\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\david\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-06-13:
Do you use antivirus tools? If yes, which? Could you try to remove `C:\Users\david\.platformio\packages\contrib-pysite` folder and restart VSCode?

davidpkgray, 2020-06-15:
Yes that was the answer, sometimes I forget about Norton antivirus working in the background. All is good now, thankyou

closed 2020-10-16 by ivankravets