platformio/platformio-vscode-ide #1792: undefinedpiogdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory

derlucas, 2020-06-14:
I just installed vscode on my Manjaro Linux using the AUR package 
visual-studio-code-bin.

VScode Version: 1.46.0
PIO core Version: 4.3.4 / Home: 3.2.3

I created a simple hello world blink project for ESP32 (esp-idf) with an olimex esp32-evb board.
I can compile and upload the code via esptool without any issues. However when i try do debug, i get this error:

> undefinedpiogdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory

My platformio.ini

> [platformio]
> default_envs = esp32-evb
> [env:esp32-evb]
> platform = espressif32
> board = esp32-evb
> framework = espidf
> upload_speed = 115200
> debug_tool = olimex-arm-usb-tiny-h
> monitor_speed = 115200

I tried olimex-arm-usb-tiny-h and segger jlink JTAG Adapters, both show the same error. I can however upload code via them when using "upload_protocol = olimex-arm-usb-tiny-h" without any problems.

I checked, and my system does not have any python2.7 installed, i have no clue where the error might originate from.

derlucas, 2020-06-14:
I think i found the source:
https://github.com/espressif/esp-idf/issues/5284#issuecomment-632754358

closed 2020-06-14 by derlucas

hpsaturn, 2022-10-13:
I had the same issue, and fixed it was install this python library. But, maybe this issue persist? because the dependency is not documented, and I have fresh installation:

`PlatformIO Core, version 6.1.4`