platformio/platformio-vscode-ide #1802: Publish extensions to open-vsx

jjchico, 2020-06-18:
Hi!

I'm a [VS Codium](https://github.com/VSCodium/vscodium) user, a FLOSS version of VS Code, it's the same code but without the Microsoft's telemetry and branding.

I want to request if possible to add the PlatformIO IDE extension on the [open-vsx.org](https://open-vsx.org/) market so that we doesn't need to works with the Microsoft marketplace as it is unclear if it's legal to use it on VS Code forks, see [here](https://github.com/microsoft/vscode/issues/31168).

This is [how to](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions) publish a new extension.

Thanks.

alfredocdmiranda, 2020-07-03:
I am with the same "problem". It would be really nice to have it published to work with VSCodium.

rafacouto, 2020-07-09:
It is technically possible:

- https://github.com/eclipse/openvsx/wiki/Publishing-Extensions
- https://github.com/VSCodium/vscodium/blob/master/DOCS.md#extensions-marketplace

@ivankravets Do you think platformio extension for codium is also legal?

ivankravets, 2020-07-09:
PlatformIO extension for VSCode depends on C/C++ extension by Microsoft. We use it for code completion/navigation. How to resolve this issue?

rafacouto, 2020-07-11:
[This one](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools&ssr=false#overview)? It is [MIT licensed](https://github.com/microsoft/vscode-cpptools).

ivankravets, 2020-07-12:
@rafacouto Yes, is it possible to have it in the open-vcs registry as well?

54696d21, 2020-07-18:
it was already merged here https://github.com/open-vsx/publish-extensions/pull/54 but it doesn't appear on open-vsx.org Probably because it doesn't build correctly? Could you help me with fixing that #1884 please?

ivankravets, 2020-07-18:
I updated this issue and added temporary solution https://github.com/platformio/platformio-vscode-ide/issues/1802

yunuscukran, 2020-10-16:
I believe the Microsoft [vscode-cpp](https://open-vsx.org/extension/vscode/cpp) extension you guys are using for code completion is [incompatible](https://github.com/open-vsx/publish-extensions/issues/86) [with](https://github.com/microsoft/vscode-cpptools/issues/5784) open-vsx.

I was able to use the [clang](https://open-vsx.org/extension/llvm-vs-code-extensions/vscode-clangd) language server instead and  getting it working (in Eclipse Theia), by

1. Deleting the ms cpp extension
2. Removing it from platformio-vscode-ide 's package.json as a dependency
3. Installing the clang extension
4.  Creating a [compile_commands.json](https://docs.platformio.org/en/latest/integration/compile_commands.html) from the PIO terminal, and putting it in the root of the project.

PlatformIO is still complaining about the incompatibility but all code completion functionality seems to work. Note that I'm using clang only as a language server, and the builds are still being handled by PIO.

This suggests to me that it is possible to separate the build and language server tools, and with small configurations, PlatformIO can actually play well with the clangd language server and therefore be open-vsx compatible.

ivankravets, 2020-10-17:
@yunuscukran Thank you so much for your solution! I added it to the top of post.

We already have started research around C/C++ extension and replacement for it. Indeed, we use noly 5% functionality of that extension - just code completion and navigation. So, we would be happy to switch to something open source with better peformance and `clang` should be a great candidate here.

ivankravets, 2020-11-03:
@yunuscukran what is your feedback about `clangd` server? Do you like how does it work?

BackMountainDevil, 2020-11-16:
> [This one](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools&ssr=false#overview)? It is [MIT licensed](https://github.com/microsoft/vscode-cpptools).

That one called vscode-cpptools is under Microsoft License, not MIT, which made this issue-[There is no a c/c++ extension on VSCodium marketplace #6518](https://github.com/microsoft/vscode-cpptools/issues/6518)

yunuscukran, 2020-11-17:
> @yunuscukran what is your feedback about `clangd` server? Do you like how does it work?

Hi,

The functionality regarding language features is good!

One aspect which I think is hacky is that the default place for` compile_commands.json` must be redefined in the project settings, and the actual act of creating the file with a PIO command must be called somewhere, via a post-build script for example. So right now clangs works with a modified PIO project.

yunuscukran, 2020-12-28:
Any updates on switching to clangd for the language server? It seems to me what is needed is simply the generation of a `compile_commands.json` file on project creation and occasional re-creations after builds.

ivankravets, 2020-12-28:
We depend on:

- https://github.com/clangd/clangd/issues/533
- https://bugs.llvm.org/show_bug.cgi?id=48266

marcdumais-work, 2021-02-03:
Hi @ivankravets

I hope you can switch to vscode-clangd ASAP - it's rapidly improving and gaining features, as is clangd. I have asked one of our contributors an update about one of your two blocking issues - I hope he can help.

I think coming together and collaborating, us non-Visual-Studio product/framework makers, open registry provider and extension consumers, to offer a better solution, vs the superficiallty-FOSS (really proprietary) extension, that we can't use anyway, is the best thing we can do.

kadircet, 2021-02-04:
Regarding clangd/clangd#533, do you people control the compiler location already? If so while waiting for clangd to have a solution around predefined macros, you can just set `--query-driver` flag yourself within the ide.

ivankravets, 2021-02-05:
@marcdumais-work thanks for the response! I welcome @valeros here and ask him to provide more details.

valeros, 2021-02-05:
Hi guys! 

@kadircet Could you please elaborate further? We're already using `--query-driver` and as you suggested in clangd/clangd#533 I opened a ticket https://bugs.llvm.org/show_bug.cgi?id=48266 as it seems to be a rather clang problem because macros generated by clang simply differ from GCC that breaks intellisense.

kadircet, 2021-02-05:
> Hi guys!
> 
> @kadircet Could you please elaborate further?

i've replied on clangd/clangd#533, but repeating it here for completeness. i was looking into both this issue and #539 at the same time. so i confused the two :/ ofc it is not possible to workaround a missing behaviour in query-driver, by enabling query-driver 😂

jerabaul29, 2021-02-07:
Looks like this is also needed to be able to use platformio from onivim2, see https://github.com/onivim/oni2/issues/3026 .

jerabaul29, 2021-03-11:
Any update on this?

I think this may be quite useful for you to release as open-vsx in order to keep users :)

For example, my main motivation for using platformio nowadays compared with Arduino IDE 2.0 is that it lets me use my favorite editor of the moment (before the Arduino IDE 2.0 came out, my main motivation for using platformio was to get linter and static analysis which was not available on Arduino 1.8, but now that Arduino IDE 2.0 is getting this and it works nicely already, this removes this part of the extra value of platformio). For me nowadays editor of choice means onivim2. So if I cannot get platformio to work for onivim2, I think that I will (unfortunately) default to switching to Arduino IDE 2.0, as I am really tired of the sluggishness of vscode and similar.

ouinouin, 2021-03-17:
Hi, 
Please release us from false opensourceness of microsoft that void many people from using the platformio extension.

cab404, 2021-04-02:
:up: C++ plugin was recently added to open-vsx. Nothing prevents deploy to @ivankravets ?

cab404, 2021-04-02:
Since C++ is present by default on VSCode anyway, maybe just drop it from deps and add popup [with download link to vsix](https://github.com/Microsoft/vscode-cpptools/releases)?

This would be a quick fix, allowing both adding PlatformIO to open-vsx and moving on with previous completion engine.

HKalbasi, 2021-04-18:
@cab404 can you do it yourself and publish a fork in open-vsx?

jjchico, 2021-06-14:
> Since C++ is present by default on VSCode anyway, maybe just drop it from deps and add popup [with download link to vsix](https://github.com/Microsoft/vscode-cpptools/releases)?
> 
> This would be a quick fix, allowing both adding PlatformIO to open-vsx and moving on with previous completion engine.

The problem, already stated above, is that license of vscode-cpptools prevents using it out of non-free MS VS Code. Suggesting the user to just circumvent the legal restriction is not appropriate.

jerabaul29, 2021-06-23:
But the idea regarding the vscode-cpptools was to instead to move to the clang tool that is fully open and permissive license, right? :) .

jankeromnes, 2021-09-29:
Hello! I'd love to help get the PlatformIO published to OpenVSX. Is there anything I can do to help? 🙂

ivankravets, 2021-09-29:
@jankeromnes Does PlatformIO IDE extension work with Theia if you install it manually? The last time when I tested it, it didn't work. Theia does not support VSCode's API on which PlatformIO IDE extension depends.

jankeromnes, 2021-09-29:
@ivankravets I'm not sure about Theia, but [Gitpod](https://github.com/gitpod-io/gitpod), [openvscode-server](https://github.com/gitpod-io/openvscode-server) and [VSCodium](https://github.com/VSCodium/vscodium) all use stock VS Code today.

ivankravets, 2021-09-29:
Wow! Founders of Theia [switched to VSCode](https://www.gitpod.io/blog/openvscode-server-launch) for Gitpod. Great decision!

I don't see [C/C++](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools) extension on Gitpod. PlatformIO IDE depends on it.

jankeromnes, 2021-09-29:
@ivankravets Yes, unfortunately `ms-vscode.cpptools` [is not open-source](https://marketplace.visualstudio.com/items/ms-vscode.cpptools/license), so it cannot be published to OpenVSX.

So, as long as `platformio.platformio-ide` has a strict dependency on it, it cannot be published to OpenVSX either.

An excellent solution discussed above would be to drop the dependency to the closed-source `ms-vscode.cpptools` extension, and replace it with a dependency to the open-source extension [vscode-clangd](https://github.com/clangd/vscode-clangd) instead. Could something like this be technically possible?

ivankravets, 2021-09-29:
@jankeromnes We would like to resolve this issue as soon as possible, but there are technical issues. Please read this topic from the beginning. For example, https://github.com/platformio/platformio-vscode-ide/issues/1802#issuecomment-751810455

sethvoltz, 2022-10-12:
@ivankravets has there been any progress on this? I am running in GitPod and the temporary workaround seemed to work, until I tried to open the Home page and it failed, as well as `pio` and `platformio` commands missing on CLI.

Zipdox2, 2023-02-14:
I second using clangd. I use clangd for all my C work in VSCodium.

qwertychouskie, 2023-05-29:
With https://github.com/platformio/platformio-vscode-ide/issues/1802#issuecomment-710683116 showing that the clang language server does work, is there any other holdup for publishing PlatformIO on <https://open-vsx.org>?

gilex-dev, 2023-10-07:
> We depend on:
> 
>     * [Query the compiler driver for pre-defined macros clangd/clangd#533](https://github.com/clangd/clangd/issues/533)
> 
>     * https://bugs.llvm.org/show_bug.cgi?id=48266

Updated link to the 2nd issue this depends on: https://github.com/llvm/llvm-project/issues/47610

alex-blueguardianco, 2024-05-04:
Any updates on this?

InfImpDrive, 2024-12-04:
Some amazing AI IDEs have come out including Windsurf, which is better than Cursor.

The pressure to be able to use PlatformIO in VSCode forks has increased because I would like AI to help speed up my Arduino-compatible projects.

InfImpDrive, 2024-12-07:
> PlatformIO extension for VSCode depends on C/C++ extension by Microsoft. We use it for code completion/navigation. How to resolve this issue?

Installing Windsurf IDE automatically copied the C/C++ Microsoft extension over.  

You have to have VSCode already installed with the extensions before you install Windsurf.

[image]

As you can see, PlatformIO is now installed, my project compiles fine, and I now have an agentic AI helping me figure out my projects for debugging and deciding what frameworks to install for my microcontroller.  I'm resisting letting it write code though, but it's a great tutor to teach me new things with Arduino!

aatanas, 2025-02-23:
Where to get the VSIX file, it seems that the dl link at https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide is removed ?

EmbeddedDevver, 2025-04-08:
It doesn't work anymore; there is a constant nagging that C/C++ tools is not allowed to be used. And then it keeps restarting every 5 minutes. Practically not workable as it starts to re-index every time.

Has anyone found a workaround?

joern-h, 2025-04-09:
@EmbeddedDevver Uninstall the vscode-cpptools extension, then reinstall it, selecting version 1.23.6 from the available versions. Also, make sure to disable automatic updates for the extension.

daniloc, 2025-07-09:
this would be very helpful for Cursor, yes.

SpeeQz1, 2025-07-10:
@ivankravets Any progress on moving over to `clang` from the `cpptools` extension?

Seen a couple of developers interested in the extension be annoyed that it can't be installed easily with VS Codium since it isn't available on the Open VSX registry.

mp035, 2025-07-31:
I just wanted to throw in a third workaround that I use on Cursor IDE.  It is very similar to [#1802](https://github.com/platformio/platformio-vscode-ide/issues/1802) but uses a different dependency.  The Cursor IDE team (Anysphere) publish their own alternative to the ms-vscode.cpptools extension on open-vsx.  The identifier is anysphere.cpptools. The steps are 

1. Download ms-vscode.cpptools and platformio.platformio-ide VSIX files using the method specified here [StackOverflow Answer](https://stackoverflow.com/a/79565372/5434939)
1.  In cursor, hit F1, type vsix and select "Extensions: Install from VSIX" and install each downloaded extension.
1. Exit Cursor IDE
1. Modify ~/.cursor/extensions/patformio.platformio-ide-X.X.X/package.json, replacing ms-vscode.cpptools with anysphere.cpptools
1. Open Cursor and uninstall msvscode.cpptools and install anysphere.cpptools
1. You will be prompted to reload the window, if not hit F1 and type "reload window", and select "Developer:  Reload Window"
1. Enjoy the amazing goodness of AI assisted embedded development. 

I have a thread on Cursor Bug Reports where I am trying to find the source and license for anysphere.cpptools.  If it is possible to publish their extension on the VSCode marketplace as well, this may be a road to a permanent solution.

davidgomes, 2025-08-17:
Hey everyone, can you please try this in Cursor? [https://open-vsx.org/extension/DavidGomes/platformio-ide](https://open-vsx.org/extension/DavidGomes/platformio-ide)

I have been able to get this to work on my local Cursor for any PlatformIO project.

What I did was:
* Make this require the Clang CPP extension (so you should get a notification to install it if you haven’t already, and this extension won’t even load until you do that)
* For every PlatformIO project, the extension now generates a compile_commands.json file in the project directory so that the Clang CPP extension works properly

mp035, 2025-10-06:
> I couldn't find any info as to why it was removed, I just noticed a few minutes ago after checking a reply to another issue.

the PlatformIO Devs asked the contributor to take it down because they were concerned that the quality of the user experience would reflect badly on the project.  I think it was very short sighted of them, at the moment the only way is to download and hack the .vsix file yourself.  That is a far worse user experience.

leomeinel, 2025-10-06:
> the PlatformIO Devs asked the contributor to take it down because they were concerned that the quality of the user experience would reflect badly on the project.

Thank you very much for letting me know. Let's just hope that one day, there will be an official version on openvsx with support for clang. :)
I can at least somewhat understand their argument though.

kevtechi, 2025-10-06:
better cursor, with notepad linting and AI prompting than nothing.

something like this: https://github.com/platformio/platformio-vscode-ide/pull/4312

mudgalp, 2026-01-16:
@davidgomes @ivankravets I know you guys are working hard to make it run on cursor ide, but I think you are surely disappointing some users by not providing anything for cursor.

or just provide a timeline so that we can wait for it quietly.

vitalNohj, 2026-02-20:
> Just for the added data point. . . Not supporting VScodium has been the standing reason I don't use PlatformIO and why I recommend others to stay away from it. Furthermore, that stance has only become stronger since the very recent betrayal of the Arduino user base by the Qualcomm buy out. Arduino IDE is totally dead now because of this. More importantly to this issue, the ability to trust is even more worn out.
> 
> Dedicating significant time and energy into any software tooling that could be pulled out from under you at the whims of a singular controlling entity is an instant deal breaker in 2026+.
> 
> The bridge has been burned. There is no more room for trust. I, like so many others, will NOT adopt any tools that aren't 100% fully and completely FOSS without the possibility or revocation. Period.
> 
> [@ivankravets](https://github.com/ivankravets). You may have thought you were being funny/snarky, but seriously... I'd rather use the CLI and a text editor as was suggested than get stuck in some kind of proprietary honey trap. That said, I can currently do exactly what you suggest with other more mature and familiar tools, making PlatformIO less useful.
> 
> PlatformIO will not see significant increased adoption or retention if this **critical issue** is not fixed.

The irony is my Cursor PIO project now IS a text editor lol No LSP for Cursor users with the hacky install. If we're doing hacky installs, the least somebody who understand the project can do is attempt a hacky fix.

InfImpDrive, 2026-02-27:
## The Compromise Workflow

For anyone looking for temporary solutions, I've also noticed multi-IDE workflows works too!

1. Create PlatformIO project in normal VSCode
2. Open the project folder in any favourite AI IDE.
3. Do your AI coding in the AI IDE
4. Do your debug/deploy in the normal VSCode IDE or heck use only CLI
5. Keep CLI mode in your build chain / Makefile
6. You can have a fully OSS toolchain "safety exit"

Now you've got a microcontroller project that can be compiled in both CLI and VSCode, while also being able to do AI agent modifications of your microcontroller project.  

You can even use your favourite AI IDE to create the correct "multiple development environment" build chain in less than 15 minutes.  Boom, now you can build at CLI without using PlatformIO.  Vibecode that makefile if you have to, and then learn how it vibecoded that.  I don't vibecode the main projects though, just pair-programmer style work.    

I just don't like to deal with the Makefile headaches, even if you do most of your manual coding and only AI-pair-program in your AI IDE.

Basically a "PlatformIO-compatible" "PlatformIO-optional" microcontroller project is quite totally feasible without the Makefile headaches of 3 years ago (thanks to AI-maintained Makefiles), so you can have cake and eat it too. 

For beautifully AI-maintained Makefiles, just make sure to keep the AI-maintained Makefile tiny and well documented with comments (so chat contexts can be reset with no problem: It instantly onboards a fresh intern of a brand new AI Chat)

My microcontroller project opens in VSCode+PlatformIO, VSCodium, Windsurf, Codeium, and Claude Code.  

It's compatible in all of that.

matthewmcneill, 2026-04-26:
Would this PR help?

https://github.com/platformio/platformio-vscode-ide/pull/4463

M