platformio/platformio-vscode-ide #1803: Problem has been occured while installing PlatformIO IDE.

lecruz, 2020-06-18:
TypeError: pioNodeHelpers.misc.runCommand is not a function at Object.runAPMCommand (file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/utils.js:60:23) at atom.notifications.addSuccess.detail (file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js:133:13) at new Promise () at file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js:132:11 at Generator.next () at step (file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js:174:2) at file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js:174:2 at new Promise () at file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js:174:2 at file:///C:/Users/j_lec/.atom/packages/platformio-ide/lib/main.js:216:52

erkapo, 2020-06-19:
TypeError: pioNodeHelpers.misc.runCommand non è una funzione in Object.runAPMCommand (file: /// C: /Users/j_lec/.atom/packages/platformio-ide/lib/utils.js: 60: 23) in atom. notifications.addSuccess.detail (file: /// C: /Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js: 133: 13) nella nuova Promise () nel file: /// C: /Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js: 132: 11 su Generator.next () al passaggio (file: /// C: /Users/j_lec/.atom /packages/platformio-ide/lib/installer/ide.js:174:2) nel file: /// C: /Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js: 174 : 2 nella nuova Promise () nel file: /// C: /Users/j_lec/.atom/packages/platformio-ide/lib/installer/ide.js: 174: 2 nel file: /// C: / Users /j_lec/.atom/packages/platformio-ide/lib/main.js:216:52

eiten, 2020-07-02:
manager.js? [sm]:83 TypeError: pioNodeHelpers.misc.runCommand is not a function
    at code-completion-engine.js? [sm]:43
    at new Promise (<anonymous>)
    at CodeCompletionEngineStage.<anonymous> (code-completion-engine.js? [sm]:43)
    at Generator.next (<anonymous>)
    at step (/home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13)
    at /home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13
    at new Promise (<anonymous>)
    at CodeCompletionEngineStage.<anonymous> (/home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13)
    at CodeCompletionEngineStage.<anonymous> (code-completion-engine.js? [sm]:11)
    at Generator.next (<anonymous>)
    at step (/home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13)
    at /home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13
    at new Promise (<anonymous>)
    at CodeCompletionEngineStage.<anonymous> (/home/edi/.atom/packages/platformio-ide/lib/installer/stages/code-completion-engine.js:13)
    at InstallationManager.<anonymous> (manager.js? [sm]:38)
    at Generator.next (<anonymous>)
    at step (/home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11)
(anonymous) @ manager.js? [sm]:83
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
Promise.then (async)
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
Promise.then (async)
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ main.js? [sm]:83
step @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ main.js? [sm]:63
step @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
activate @ main.js? [sm]:57
activateNow @ <embedded>:11
(anonymous) @ <embedded>:14
simpleDispatch @ <embedded>:11
emit @ <embedded>:11
(anonymous) @ <embedded>:11
activatePackage @ <embedded>:11
(anonymous) @ <embedded>:11
r @ <embedded>:11
n @ <embedded>:14
(anonymous) @ <embedded>:14
(anonymous) @ <embedded>:14
emit @ events.js:199
(anonymous) @ net.js:593
manager.js? [sm]:83 Error: Missed packages: build,busy-signal,platformio-ide-terminal,platformio-ide-debugger,autocomplete-clang,linter-ui-default,intentions,linter,linter-gcc2,language-ini,tool-bar,file-icons
    at AtomDependenciesStage.check (atom-dependencies.js? [sm]:129)
    at InstallationManager.<anonymous> (manager.js? [sm]:38)
    at Generator.throw (<anonymous>)
    at step (/home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11)
(anonymous) @ manager.js? [sm]:83
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
Promise.then (async)
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
Promise.then (async)
step @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/installer/manager.js:11
(anonymous) @ main.js? [sm]:83
step @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ main.js? [sm]:63
step @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
(anonymous) @ /home/edi/.atom/packages/platformio-ide/lib/main.js:11
activate @ main.js? [sm]:57
activateNow @ <embedded>:11
(anonymous) @ <embedded>:14
simpleDispatch @ <embedded>:11
emit @ <embedded>:11
(anonymous) @ <embedded>:11
activatePackage @ <embedded>:11
(anonymous) @ <embedded>:11
r @ <embedded>:11
n @ <embedded>:14
(anonymous) @ <embedded>:14
(anonymous) @ <embedded>:14
emit @ events.js:199
(anonymous) @ net.js:593
manager.js? [sm]:83 Error: PlatformIO Core has not been installed yet!
    at t.value (/home/edi/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1)

closed 2020-09-18 by stale[bot]