platformio/platformio-vscode-ide #1805: Start PIO Home Server

ImQuiteBored, 2020-06-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.46.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "c:\users\[user]\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "c:\users\[user]\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\[user]\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\[user]\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\[user]\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\[user]\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\[user]\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\[user]\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 40, in <module>
    from autobahn.twisted.util import sleep
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\autobahn\twisted\util.py", line 31, in <module>
    from twisted.internet.defer import Deferred
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\internet\defer.py", line 32, in <module>
    from twisted.logger import Logger
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 102, in <module>
    from ._logger import Logger, _loggerFor
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\logger\_logger.py", line 274, in <module>
    _log = Logger()
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\logger\_logger.py", line 70, in __init__
    from ._global import globalLogPublisher
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\logger\_global.py", line 16, in <module>
    from ._buffer import LimitedHistoryLogObserver
  File "C:\Users\[user]\.platformio\packages\contrib-pysite\twisted\logger\_buffer.py", line 11, in <module>
    from zope.interface import implementer
ModuleNotFoundError: No module named 'zope'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\[user]\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\[user]\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\[user]\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

jagimene, 2020-08-04:
Hi, i have the same issue. Did you find a solution?

ImQuiteBored, 2020-08-06:
@jagimene it was an issue with the main branch of Marlin. I had to switch to the current bugfix version.

closed 2020-09-21 by stale[bot]