platformio/platformio-vscode-ide #1827: PlatformIO Building freezes and does nothing.

zekageri, 2020-06-26:
### When i try to build a project with esp32 PIO is just hangs and does nothing. 
**This is not a freeze. Its different.**

### My PIO.ini:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

upload_protocol = espota
upload_port = 192.168.0.33
monitor_speed = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB (128mb)
board_build.flash_mode = dio
board_build.partitions = large_spiffs_16MB.csv
```

**I have tried to update all PIO related stuffs,  deleted .vscode folder, restarted pc.**

zekageri, 2020-06-26:
Other Projects are building correctly and uploading fine.

zekageri, 2020-06-26:
If i create a new project, and copy the src and the lib and the pio.ini to the new project from the old, the issue is the same.

zekageri, 2020-06-29:
https://github.com/platformio/platformio-core/issues/3417#issuecomment-601498541

closed 2020-06-29 by zekageri

tjhaas17, 2021-02-14:
Have you figured this out? I'm having the same problem and cant seem to find any other sources

YiffyToys, 2021-03-03:
I seem to have the same issue "byte 0x81 in position 8" in issue #2405.
It auto update to the latest version.
`You're up-to-date!
PlatformIO 5.1.0 is currently the newest version available.`

So this issue does NOT seem to be fixed.